dtcooper/raspotify #1: USB Soundcard

noiprcocks, 2017-06-08:
Hey mate,

can't get it working with my usb soundcard. Did like exactly like mentioned in the readme with
 "`--device hw:0,1".
Even tried "output = alsasink device=hw:1,0". But no success. Whatever I do, the Music Plays through the hdmi out ...
Got mopidy running too. And there it works fine and like expected: Sound Comes through the 3.5mm jack of the usb soundcard.
Any tip would be awesome.

dtcooper, 2017-06-08:
Hmmm, haven't actually tested this with a USB sound device. I actually have one on hand so I'll give it a try soon and report back.

noiprcocks, 2017-06-08:
Awesome! Let me know if I can provide you with any logs or additonal info.
I'm using the google home voice assist so sadly im forced to use my usb Card.

dtcooper, 2017-06-09:
Hey @noiprcocks 

What USB sound card are you using? I confirm it working via the following with my [Griffin iMic](https://griffintechnology.com/us/imic) USB dongle,

```
$ lsusb
Bus 001 Device 004: ID 077d:07af Griffin Technology iMic
### ...Ignore some lines

$ aplay -l
### ...Ignore some lines
card 1: system [iMic USB audio system], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
```

For me `card 1: system [iMic USB audio system], device 0: USB Audio [USB Audio]` denotes ALSA device `hw:1,0`, so I add the following to `/etc/default/raspotify`,

```
# /etc/default/raspotify

DEVICE_NAME="USB Dongle Test"
OPTIONS="--username 1234567890 --password mypassword --device hw:1,0"
```

And then I restart the service via `sudo systemctl restart raspotify`. It shows up in Spotify connect. (...Even when I'm not logged on to the local network because I provided my username/password!)

Are you sure your USB device is working in the first place? Note you may have to get your USB device working via the command `alsamixer`. Try getting it working using [SoX](http://sox.sourceforge.net/) first, at the command line,

```
### Configure/unmute the soundcard with alsamixer, you'll have to press F6 to select the card
$ alsamixer

### Install SoX, which is an awesome command-line audio player known to work
$ sudo apt-get -y install sox libsox-fmt-all

### Set AUDIODEV to whatever you believe would be the sound card based on
### "aplay -l" as shown above and play my radio station, 820hz :P
$ AUDIODEV=hw:1,0 play 'http://radio.820hz.fm/main.mp3'
```

Does that help? Are you able to get it working with SoX first?

Thanks,

David

noiprcocks, 2017-06-10:
Wasn't seeing the wood for the trees... i was messing up the 

`OPTIONS="--username 1234567890 --password mypassword --device hw:1,0"`

I missed the second "-" before device. 

Thanks a lot.

closed 2017-06-10 by noiprcocks