dtcooper/raspotify #2: Can this be used to cast to another spotify client/chromecast etc?

kristoffernolgren, 2017-06-09:
Really cool project! When reading the docs, it sounds like this implementation is built to use as a reciever to a speaker only. I'm looking for something that I can pair with my home-integration-system that can initiate a casting to a wireless speaker. This functionality exist in spotify already, but I don't want to be dependent on a cellphone with an app.

dtcooper, 2017-06-09:
I'm not really sure I follow. I've never used a Chromecast. I doubt this feature is supported upstream by [librespot](https://github.com/plietar/librespot), but you can try your luck there?

closed 2017-06-09 by dtcooper

kristoffernolgren, 2017-06-10:
If you go into spotify, you can select what instance of spotify you want to use as output, where you select your devices. In this list, intaces of spotify and supported devices show up. Sonos-speakers, chromecasts etc.

I want to use raspotify to cast to another source via commandline. So i think my question is twofold. 1) Can I run my own scripts such as librespot/examples/play.rs using this project or is it more of a one-off thing as a thin client? If that's possible, how do you tell raspotify what script to run?
2) Does raspotify/librespot support playing on other devices?

cortegedusage, 2017-06-10:
Maybe this is more what you seek:
https://github.com/Spotifyd/spotifyd-http

cortegedusage, 2017-06-10:
With this you can control other clients

kristoffernolgren, 2017-06-10:
Thankyou! The problem with the spotify connect API is that it only works if you have used another player to establish the connection with. If you quit the app on the receiving end or start casting something else to a speaker, it seems to disapear from the /devices – endpoint.