dtcooper/raspotify #3: Stops playing after x songs

higab85, 2017-06-28:
Hey! 
So let me start by saying this the most convenient solution to use spotify connect on an rpi, thank you!! 

The issue I've run into, is that it stops playing after x tracks. So it will finish playing the track, and then stop until I `service raspotify restart` . Here's what I get for `service raspotify status` :

```
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled)
   Active: active (running) since Wed 2017-06-28 11:05:01 UTC; 20min ago
  Process: 1670 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 1667 ExecStartPre=/bin/mkdir -p 0755 /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 1673 (librespot)
   CGroup: /system.slice/raspotify.service
           ├─1673 /usr/bin/librespot --name raspotify (raspberrypi) --backend alsa --cache /var/cache/raspotify --bitrate 160
           └─1686 /usr/bin/pulseaudio --start --log-target=syslog

Jun 28 11:05:10 raspberrypi librespot[1673]: INFO:librespot::player: Track "A Jealous Heart Never Rests" loaded
Jun 28 11:05:11 raspberrypi pulseaudio[1686]: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Jun 28 11:05:11 raspberrypi pulseaudio[1686]: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Jun 28 11:05:11 raspberrypi pulseaudio[1686]: [pulseaudio] bluez4-util.c: org.bluez.Manager.GetProperties() failed: org.freedesktop.DBus.Error.UnknownMethod: Method "GetProperties" with signature "" on interface "org.bluez.Manager" doesn't exist
Jun 28 11:06:31 raspberrypi librespot[1673]: INFO:librespot::player: Loading track "Playing with Fire"
Jun 28 11:06:32 raspberrypi librespot[1673]: INFO:librespot::player: Track "Playing with Fire" loaded
Jun 28 11:06:35 raspberrypi librespot[1673]: INFO:librespot::player: Loading track "Friday Fish Fry - Maribou State & Pedestrian Remix - Radio Edit"
Jun 28 11:06:35 raspberrypi librespot[1673]: INFO:librespot::player: Track "Friday Fish Fry - Maribou State & Pedestrian Remix - Radio Edit" loaded
Jun 28 11:10:46 raspberrypi librespot[1673]: INFO:librespot::player: Loading track "Waves"
Jun 28 11:10:46 raspberrypi librespot[1673]: INFO:librespot::player: Track "Waves" loaded
```

staubrein, 2017-07-02:
I have the same issue!
Commenting everything in the config file worked for one track,
after that it randomly switched to another song and stopped playing after 3 seconds.
Cleared the cache - nothing.

My status logs seems harmless:
`pi@raspberrypi:~ $ sudo service raspotify status
● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled)
   Active: active (running) since Sun 2017-07-02 14:54:22 UTC; 4s ago
  Process: 3984 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 3981 ExecStartPre=/bin/mkdir -p 0755 /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 3987 (librespot)
   CGroup: /system.slice/raspotify.service
           └─3987 /usr/bin/librespot --name raspotify (raspberrypi) --backend alsa --cache /var/cache/raspotify --bitrate 160

Jul 02 14:54:22 raspberrypi systemd[1]: Starting Raspotify...
Jul 02 14:54:22 raspberrypi systemd[1]: Started Raspotify.
Jul 02 14:54:22 raspberrypi librespot[3987]: INFO:librespot: librespot 2259188 (2017-06-14). Built on 2017-06-14.
Jul 02 14:54:23 raspberrypi librespot[3987]: INFO:librespot::session: Connecting to AP "lon3-accesspoint-a3.ap.spotify.com:4070"
Jul 02 14:54:24 raspberrypi librespot[3987]: INFO:librespot::session: Authenticated as "xxxxx" !
Jul 02 14:54:24 raspberrypi librespot[3987]: INFO:librespot::session: Country: "XX"
Jul 02 14:54:24 raspberrypi librespot[3987]: INFO:librespot::audio_backend::alsa: Using alsa sink
`

dtcooper, 2017-07-03:
I can't seem to get a repro, guys. What happens if you try a couple songs using the `librespot` command without the daemon, in verbose mode?

```
$ librespot --name "librespot test" --backend alsa --cache /tmp/librespot-test --bitrate 160 --verbose
```

staubrein, 2017-07-04:
Tested it, same issue.
It starts to play a song and drops it after a few seconds.

Here the interesting part of the debugging information:
```
TRACE:librespot::audio_file: chunk 7 / 23 complete
TRACE:librespot::audio_file: requesting chunk 8
TRACE:librespot::audio_file: chunk 8 / 23 complete
TRACE:librespot::audio_file: requesting chunk 9
TRACE:librespot::audio_file: chunk 9 / 23 complete
TRACE:librespot::audio_file: requesting chunk 10
TRACE:librespot::audio_file: chunk 10 / 23 complete
TRACE:librespot::audio_file: requesting chunk 11
TRACE:librespot::audio_file: chunk 11 / 23 complete
TRACE:librespot::audio_file: requesting chunk 12
TRACE:librespot::audio_file: chunk 12 / 23 complete
TRACE:librespot::audio_file: requesting chunk 13
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Error { repr: Os { code: 28, message: "No space left on device" } }', /checkout/src/libcore/result.rs:859
note: Run with `RUST_BACKTRACE=1` for a backtrace.
DEBUG:librespot::spirc: drop Spirc[0]
DEBUG:librespot::session: drop Dispatch
```

cortegedusage, 2017-07-05:
Looks like 
https://github.com/plietar/librespot/issues/166
Might want to disable audio cache, i believe this is an option in latest release.

staubrein, 2017-07-05:
In my case it's an Pi Model B+ with only raspotify running so there should be enough RAM.

derWeihnachtsmann, 2017-07-05:
Then test it urself. open 2 SSH seesions, one for librespot and one for #watch -n0,5 free -h
and let them play. You can also watch df -h for cache on storage. I believe the cache is not in RAM but in storage. (temp or so)

dtcooper, 2017-07-05:
Ah, I'll add the new argument to avoid caching! Hopefully that will work.

D

dtcooper, 2017-07-05:
@higab85, @staubrein et al., you might like to try the latest version. Install by,

```
curl -sL https://raw.github.com/dtcooper/raspotify/master/install.sh | sh
```

dtcooper, 2017-07-05:
(Have potential fix out in version [0.7](https://github.com/dtcooper/raspotify/releases/tag/0.7).)

lovicub, 2017-07-05:
perfect. i am listening to music since 20 minutes and everything goes properly :)
hmmmm... i've noticed faster respond time, and you guys?

dtcooper, 2017-07-06:
@lovicub great! Closing this issue. If similar problems crop up, please reopen. Thanks!

millocorona, 2017-07-16:
I'm still having the same problem even with all the config file commented 

`[pi@Coruscant ~]$ sudo systemctl status raspotify -l
● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled)
   Active: active (running) since Sun 2017-07-16 15:21:48 CDT; 6min ago
  Process: 8975 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 8971 ExecStartPre=/bin/mkdir -p 0755 /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 8978 (librespot)
   CGroup: /system.slice/raspotify.service
           ├─8978 /usr/bin/librespot --name raspotify (Coruscant) --backend alsa --bitrate 160 --disable-audio-cache
           └─9001 /usr/bin/pulseaudio --start --log-target=syslog

Jul 16 15:25:29 Coruscant librespot[8978]: INFO:librespot::player: Track "Guiding Light" loaded
Jul 16 15:25:30 Coruscant pulseaudio[9001]: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Jul 16 15:25:30 Coruscant pulseaudio[9001]: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Jul 16 15:25:30 Coruscant pulseaudio[9001]: [pulseaudio] bluez4-util.c: org.bluez.Manager.GetProperties() failed: org.freedesktop.DBus.Error.UnknownMethod: Method "GetProperties" with signature "" on interface "org.bluez.Manager" doesn't exist
Jul 16 15:25:46 Coruscant librespot[8978]: INFO:librespot::player: Loading track "Guiding Light"
Jul 16 15:25:47 Coruscant librespot[8978]: INFO:librespot::player: Track "Guiding Light" loaded
Jul 16 15:25:53 Coruscant librespot[8978]: INFO:librespot::player: Loading track "Enter Sandman"
Jul 16 15:25:55 Coruscant librespot[8978]: INFO:librespot::player: Track "Enter Sandman" loaded
Jul 16 15:26:22 Coruscant librespot[8978]: INFO:librespot::player: Loading track "Animals"
Jul 16 15:26:23 Coruscant librespot[8978]: INFO:librespot::player: Track "Animals" loaded
`
thats the status of systemctl, keeps running but it doesnt play anything

kreague, 2017-08-12:
Hi, I just wanted to leave some feedback that is hopefully helpful. 

I was having this exact issue on a very recent deployment off the default raspbian image on a first gen pi with the latest raspotify installed (installed about a week ago). 

Following some advice I found on a different project, I uninstalled pulseaudio, and have not had the issue since.  Works perfectly now, at least for the last 48 hours anyway. YMMV

dtcooper, 2017-08-12:
Ah, I've only tested on Raspbian Jessie Lite, which doesn't have Pulse.

markheloking, 2017-10-04:
For me audio stops after about 3 songs and I'm not sure why. I disabled audio cache, but it's still happening. On my Android Spotify client it still tells me the song is playing and if I request the service status it tells me the song is loaded and yet no audio.
Any more suggestions? How did you guys fix this?

fmarmar, 2017-12-18:
I have the same exact problem. I even can fast-forward the problem skipping the current song to the next in my player. 

The 4th song is never played, and following songs don't even appear in the logs.

@markheloking the behaviour you describe happens every time and it doesn't seem to be the problem. 

Here are my logs:
[librespot.log](https://github.com/dtcooper/raspotify/files/1569626/librespot.log)

fmarmar, 2017-12-19:
Hi again, 

I can confirm @kreague solution. The problem is the interaction with pulseaudio.

Sereniity, 2017-12-22:
@kreague @fmarmar Been having this issue, removed pulseaudio also and seems to be fixed. Mine always failed on the 3rd song, up to 10 now and still playing fine.

Giraffenmatte, 2018-01-18:
@kreague @fmarmar I had the same issue, removed pulseaudio and now it works. Thanks so much.

andreipopovici, 2018-01-28:
What about the rest of us poor souls who absolutely have to use `pulseaudio`? Any chance to get this issue fixed?

justintgav, 2018-02-11:
In the same boat as @andreipopovici , need to keep pulseaudio around. Any suggestions? Seeing the same  problem: dies after one song.

andreipopovici, 2018-02-11:
FYI, I ended up giving up on `pulseaudio`. It seems the newer version of `dietpi` I'm using has better `dmix` support, so I'm using that.

niko-tin, 2018-03-17:
I had the same problem using Ubuntu Mate on my RPi 3B. After 3 songs the sound stopped but Spotify was telling me everything is still playing. After I followed the instructions 

> Here is what you have to do:
>
>change to your desktop user and replace <user> and <group> with yours, in my case it's pi
>su <user>
>cd ~/
>
>copy the global config file to your user directory
>mkdir ~/.pulse/
>sudo cp /etc/pulse/default.pa ~/.pulse/
>
>dd this line to the end of the file you've just copied
>sudo nano ~/.pulse/default.pa
>load-module module-native-protocol-tcp auth-ip-acl=127.0.0.1
>sudo chown -R $USER ~/.pulse
>
>go to the raspotify user home dir
>cd /var/cache/raspotify
>sudo mkdir .pulse
>sudo sh -c 'echo "default-server = 127.0.0.1" > .pulse/client.conf'
>sudo chown -R raspotify:raspotify ~/.pulse

>reboot and enjoy

from  #9 - there were no sound at all. So I changed everything back. Now I have no problems playing songs via spotify and everything works fine. Even after restarting the Pi

kwinz, 2018-04-12:
Really annoying issue. Kind of defeats the whole purpose of raspotify.

I don't have `/etc/pulse/default.pa` so I can't try the fix @niko-tin

dtcooper, 2018-04-12:
Can anybody give me exact STR, or perhaps what pulse related packages you have installed?. Thanks.

abrewner, 2019-09-05:
> Following some advice I found on a different project, I uninstalled pulseaudio, and have not had the issue since. Works perfectly now, at least for the last 48 hours anyway. YMMV

Simply uninstalling pulseaudio worked for me as well. Running raspotify 0.13.1 of raspbian stretch 9.3.
Although the sound stops when I skim through a song, but the next one get's played normally again.

benevid, 2020-10-23:
> Hi again,
> 
> I can confirm @kreague solution. The problem is the interaction with pulseaudio.

I am use Pi 4 (raspian-Magic Mirror) with the lastest (0.14) and have experienced the same problem. Fixed it removing pulseaudio!

But, pulseaudio is important for?

jukisu23, 2021-03-16:
I don't have pulseaudio installed. Still the same issue: stops playing after approx. 9-10 minutes every time (MID SONG).
I checked and it doesn't engage the screensaver (which some pointed out).

My log shows

```
  [2021-03-16T18:02:45Z ERROR librespot_playback::player] Unable to load audio item.
  [2021-03-16T18:02:45Z ERROR librespot_core::session] Os { code: 104, kind: ConnectionReset, message: "Connection reset by peer" }
  [2021-03-16T18:02:45Z WARN  librespot] Spirc shut down unexpectedly
  [2021-03-16T18:02:45Z INFO  librespot_core::session] Connecting to AP "gew1-accesspoint-e-n595.ap.spotify.com:4070"
  [2021-03-16T18:02:45Z INFO  librespot_core::session] Authenticated as "l50x5fzoht9n2ivpvcqf3pnqn" !
  [2021-03-16T18:02:45Z INFO  librespot_playback::audio_backend::alsa] Using alsa sink
  [2021-03-16T18:02:45Z INFO  librespot_core::session] Country: "DE"

```

Any ideas?

jukisu23, 2021-03-16:
This is the output of the raw librespot:
```

2021-03-16T18:37:20Z DEBUG librespot_playback::player] command=Preload(SpotifyId { id: 1068282439110763866323352787392203638, audio_type: Track })
[2021-03-16T18:37:20Z DEBUG librespot_playback::player] Preloading track
[2021-03-16T18:37:20Z INFO  librespot_playback::player] Loading <Come Over Love> with Spotify URI <spotify:track:01w1yDztuPjNHkTmXXtIP4>
[2021-03-16T18:37:20Z DEBUG librespot_audio::fetch] Downloading file 441823b68db51ed2850faa3a8ca5977f9af1f8ad
[2021-03-16T18:37:20Z TRACE librespot_audio::fetch] Ping time estimated as: 107 ms.
[2021-03-16T18:37:20Z DEBUG librespot_playback::player] Normalisation Data: NormalisationData { track_gain_db: -10.695622, track_peak: 0.97112167, album_gain_db: -10.695622, album_peak: 0.97112167 }
[2021-03-16T18:37:20Z DEBUG librespot_playback::player] Applied normalisation factor: 0.2918898
[2021-03-16T18:37:20Z INFO  librespot_playback::player] <Come Over Love> (208280 ms) loaded
[2021-03-16T18:37:20Z DEBUG librespot_core::session] Invalidating session[0]
[2021-03-16T18:37:20Z DEBUG librespot_core::session] drop Dispatch
[2021-03-16T18:37:20Z ERROR librespot_core::session] Os { code: 104, kind: ConnectionReset, message: "Connection reset by peer" }
[2021-03-16T18:37:20Z WARN  librespot] Spirc shut down unexpectedly
[2021-03-16T18:37:20Z DEBUG librespot_connect::spirc] drop Spirc[0]
[2021-03-16T18:37:20Z DEBUG librespot_playback::player] Shutting down player thread ...
[2021-03-16T18:37:20Z DEBUG librespot_playback::player] drop PlayerInternal[0]
[2021-03-16T18:37:20Z DEBUG librespot_playback::player] PlayerInternal thread finished.
[2021-03-16T18:37:20Z WARN  librespot_audio::fetch] Error from channel for data receiver for range 167936 (+16384).
[2021-03-16T18:37:20Z DEBUG librespot_core::session] drop Session[0]
[2021-03-16T18:37:20Z DEBUG librespot_core::audio_key] drop AudioKeyManager
[2021-03-16T18:37:20Z DEBUG librespot_core::channel] drop ChannelManager
[2021-03-16T18:37:20Z DEBUG librespot_core::mercury] drop MercuryManager
[2021-03-16T18:37:21Z INFO  librespot_core::session] Connecting to AP "gew1-accesspoint-e-td73.ap.spotify.com:4070"
[2021-03-16T18:37:21Z INFO  librespot_core::session] Authenticated as "_____________________" !
[2021-03-16T18:37:21Z DEBUG librespot_core::session] new Session[1]
[2021-03-16T18:37:21Z DEBUG librespot_connect::spirc] new Spirc[1]
[2021-03-16T18:37:21Z DEBUG librespot_playback::player] new Player[1]
[2021-03-16T18:37:21Z INFO  librespot_playback::audio_backend::alsa] Using alsa sink
[2021-03-16T18:37:21Z DEBUG librespot_connect::spirc] canonical_username: l50x5fzoht9n2ivpvcqf3pnqn
[2021-03-16T18:37:21Z DEBUG librespot_core::mercury] new MercuryManager
[2021-03-16T18:37:21Z DEBUG librespot_connect::spirc] input volume:32768 to mixer: 2072
[2021-03-16T18:37:21Z DEBUG librespot_playback::player] command=AddEventSender
[2021-03-16T18:37:21Z DEBUG librespot_playback::player] command=VolumeSet(32768)
[2021-03-16T18:37:21Z DEBUG librespot_core::session] Session[1] strong=3 weak=2
[2021-03-16T18:37:21Z INFO  librespot_core::session] Country: "DE"
[2021-03-16T18:37:21Z DEBUG librespot_core::mercury] unknown subscription uri=hm://remote/user/l50x5fzoht9n2ivpvcqf3pnqn/
[2021-03-16T18:37:21Z DEBUG librespot_core::mercury] subscribed uri=hm://remote/user/l50x5fzoht9n2ivpvcqf3pnqn/ count=0
[2021-03-16T18:37:22Z DEBUG librespot_connect::spirc] kMessageTypeNotify "jp42" 2ec60a76fd718d78711937e802bdca1ab1265523 1012139529 1615919842073 kPlayStatusStop
[2021-03-16T18:37:22Z DEBUG librespot_connect::spirc] kMessageTypeNotify "jp42" 2ec60a76fd718d78711937e802bdca1ab1265523 1012139768 1615919842312 kPlayStatusStop

```

JasonLG1979, 2021-11-16:
Closing because of age and inactivity. A new version of Raspotify based on a much newer version librespot will be released soon. When that happens please update and if the issue persists reopen the issue.

closed 2021-11-16 by JasonLG1979