dtcooper/raspotify #5: Custom backend

swilalaa, 2017-07-07:
Even if add in /etc/default/raspotify to change --backend and --device raspotify still use alsa as backend.

Any tip where I can change that? 

I would like to add those argues:

--backend pipe --device /home/pi/srv/music/spotify --cache /tmp > /dev/null 2>&1

dtcooper, 2017-07-07:
I'm not sure I follow the question. You can edit the systemd service to run librespot with whatever arguments you'd like. I think pipe mode would be unsupported.

swilalaa, 2017-07-08:
I edited the systemd service directly, and added the arguments I want, the service will start but the connect client is visible but noch reachable.

When I put the same command directly into the commandline it works - also with pipe mode.

swilalaa, 2017-07-08:
Found the issue, the service was running under a different user as my pipe did.

closed 2017-07-08 by swilalaa