dtcooper/raspotify #7: Building a more generic Debian librespot package

mfonville, 2017-07-21:
would it maybe be possible to further enhance your scripts to e.g. build packages (and upstream the scripts) so that librespot could be packaged for the regular Debian universe repositories?

dtcooper, 2017-07-21:
Sure, why not @mfonville. I've never packaged anything for Debian beyond this. Any pointers about where to start?

mfonville, 2017-07-24:
I think best is to start with https://wiki.debian.org/Teams/RustPackaging/Policy and use `debcargo` or `dhcargo` to package the app.

Building can be done with e.g. pbuilder, or debuild with a Ubuntu PPA at first, and when happy with the results, applying for inclusion to the Debian universe maintainers.

If having made the debian templates, they could also be upstreamed to librespot.

dtcooper, 2017-12-27:
Labeling this as wontfix, since the intended audience is Pi users.

closed 2017-12-27 by dtcooper

Rmannn, 2018-01-07:
But pi users may use classics debian...