dtcooper/raspotify #8: Tracks auto repeating despite repeat being turned off

Logarythms, 2017-07-25:
So I've been using Raspotify for a few days and so far it's been working great, with one minor issue:

The playback behaves as if the repeat setting is turned on, despite it actually being turned off. So if I'm listening to an album and the last song finishes, the first track of that album follows immediately after. Turning the repeat mode on and off doesn't change anything.

Additionally, the repeat-track mode seems to be missing completely.

michalfita, 2017-09-03:
It's even worse, as there is no control over looping - a very important feature for me as I'm using it for lullabies for my baby.

JasonLG1979, 2021-11-16:
Closing because of age and inactivity. A new version of Raspotify based on a much newer version librespot will be released soon. When that happens please update and if the issue persists reopen the issue.

closed 2021-11-16 by JasonLG1979