dtcooper/raspotify #9: Streaming with no sound

maciekbaczmanski, 2017-08-13:
Sooo. I've installed raspotify and it works like a charm.... but only for first 5 songs. After that I can still stream on spotify connect (my spotify says the song is playing) but I can't hear anything... When I restart I can again listen to something like 5 songs... Any ideas?

dtcooper, 2017-08-13:
Quick fix appears to be uninstalling PulseAudio ([see this comment](https://github.com/dtcooper/raspotify/issues/3#issuecomment-321962287)). Since I've never used Raspotify (librespot) with just Pulse, I'll try to test it soon and see if I can come up with a fix.

maciekbaczmanski, 2017-08-14:
After uninstalling PulseAudio It doesn't even play at least one song... :/ Tried reinstaling it, but without success

maciekbaczmanski, 2017-08-14:
This is what I get while trying to play music:
pi@raspberrypi:~ $ sudo systemctl status raspotify -l
● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled)
   Active: active (running) since Mon 2017-08-14 10:47:46 UTC; 1min 46s ago
  Process: 1379 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 1376 ExecStartPre=/bin/mkdir -p 0755 /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 1382 (librespot)
   CGroup: /system.slice/raspotify.service
           └─1382 /usr/bin/librespot --name raspotify (raspberrypi) --backend alsa --bitrate 160 --disable-audio-cache

Aug 14 10:47:46 raspberrypi librespot[1382]: WARN:mdns::fsm: error sending packet Error { repr: Os { code: 99, message: "Cannot assign requested address" } }
Aug 14 10:49:23 raspberrypi librespot[1382]: INFO:librespot_core::session: Connecting to AP "gew1-accesspoint-b-f843.ap.spotify.com:4070"
Aug 14 10:49:23 raspberrypi librespot[1382]: INFO:librespot_core::session: Authenticated as "maciekbaczmańsi" !
Aug 14 10:49:23 raspberrypi librespot[1382]: INFO:librespot::audio_backend::alsa: Using alsa sink
Aug 14 10:49:23 raspberrypi librespot[1382]: INFO:librespot_core::session: Country: "PL"
Aug 14 10:49:24 raspberrypi librespot[1382]: INFO:librespot::player: Loading track "OK"
Aug 14 10:49:25 raspberrypi librespot[1382]: INFO:librespot::player: Track "OK" loaded
Aug 14 10:49:25 raspberrypi librespot[1382]: ALSA lib pcm_hw.c:1667:(_snd_pcm_hw_open) Invalid value for card
Aug 14 10:49:25 raspberrypi librespot[1382]: thread '<unnamed>' panicked at 'called `Option::unwrap()` on a `None` value', /checkout/src/libcore/option.rs:335
Aug 14 10:49:25 raspberrypi librespot[1382]: note: Run with `RUST_BACKTRACE=1` for a backtrace.

tomtana, 2017-08-14:
I can confirm that behavior. After 5 songs the librespot library seems to crash. It doesnt matter if the songs are skipped or fully played. Here a verbose output where some lines are left out for convenience. 

```

Aug 14 11:00:24 rasperry librespot[3490]: INFO:librespot: librespot ddfc28f (2017-08-04). Built on 2017-08-07. Build ID: N5RcEuis
Aug 14 11:00:46 rasperry librespot[3490]: DEBUG:librespot::discovery: Post "/" {}
Aug 14 11:00:46 rasperry librespot[3490]: INFO:librespot_core::session: Connecting to AP "lon6-accesspoint-a30.ap.spotify.com:4070"
Aug 14 11:00:47 rasperry librespot[3490]: INFO:librespot_core::session: Authenticated as "*****" !
Aug 14 11:00:47 rasperry librespot[3490]: DEBUG:librespot_core::session: new Session[0]
Aug 14 11:00:47 rasperry librespot[3490]: DEBUG:librespot::spirc: new Spirc[0]
Aug 14 11:00:47 rasperry librespot[3490]: DEBUG:librespot_core::mercury: new MercuryManager
Aug 14 11:00:47 rasperry librespot[3490]: DEBUG:librespot::player: new Player[0]
Aug 14 11:00:47 rasperry librespot[3490]: INFO:librespot::audio_backend::alsa: Using alsa sink
Aug 14 11:00:47 rasperry librespot[3490]: DEBUG:librespot_core::session: Session[0] strong=3 weak=2
Aug 14 11:00:47 rasperry librespot[3490]: INFO:librespot_core::session: Country: "DE"
Aug 14 11:00:47 rasperry librespot[3490]: DEBUG:librespot_core::mercury: unknown subscription uri=hm://remote/3/user/tomtana/2cb0f80d51724694a8409b54290eec7963edd616
Aug 14 11:00:47 rasperry librespot[3490]: DEBUG:librespot_core::mercury: subscribed uri=hm://remote/3/user/*****/ count=0
Aug 14 11:00:47 rasperry librespot[3490]: DEBUG:librespot::spirc: kMessageTypeNotify "GT-I9195I" 13b432bc1352df2454869a740835594f211e6e7d 1 1502708431284
Aug 14 11:00:47 rasperry librespot[3490]: DEBUG:librespot::spirc: kMessageTypeLoad "GT-I9195I" 13b432bc1352df2454869a740835594f211e6e7d 2 0
Aug 14 11:00:47 rasperry librespot[3490]: DEBUG:librespot::player: command=Load(SpotifyId(u128 { high: 16983298126435402182, low: 11493134635707380817 }), false, 14598)
Aug 14 11:00:47 rasperry librespot[3490]: DEBUG:librespot::spirc: kMessageTypeNotify "GT-I9195I" 13b432bc1352df2454869a740835594f211e6e7d 138 1502708431284
Aug 14 11:00:47 rasperry librespot[3490]: INFO:librespot::player: Loading track "I Can See Clearly Now"
Aug 14 11:00:47 rasperry librespot[3490]: DEBUG:librespot_core::audio_key: new AudioKeyManager
Aug 14 11:00:47 rasperry librespot[3490]: DEBUG:librespot::spirc: kMessageTypeNotify "raspotify (rasperry.com)" 2cb0f80d51724694a8409b54290eec7963edd616 2 1502708447682
Aug 14 11:00:47 rasperry librespot[3490]: DEBUG:librespot_audio::fetch: Downloading file 3dbcb2e7557c4ce1cca3893a6ba52656217dbab2
Aug 14 11:00:47 rasperry librespot[3490]: TRACE:librespot_audio::fetch: requesting chunk 0
Aug 14 11:00:47 rasperry librespot[3490]: DEBUG:librespot_core::channel: new ChannelManager
Aug 14 11:00:48 rasperry librespot[3490]: TRACE:librespot_audio::fetch: chunk 0 / 30 complete
....
Aug 14 11:00:48 rasperry librespot[3490]: INFO:librespot::player: Track "I Can See Clearly Now" loaded
.....
Aug 14 11:00:49 rasperry librespot[3490]: DEBUG:librespot::spirc: kMessageTypePlay "GT-I9195I" 13b432bc1352df2454869a740835594f211e6e7d 3 0
Aug 14 11:00:49 rasperry librespot[3490]: DEBUG:librespot::player: command=Play
Aug 14 11:00:49 rasperry librespot[3490]: TRACE:librespot_audio::fetch: chunk 9 / 30 complete
Aug 14 11:00:49 rasperry librespot[3490]: TRACE:librespot_audio::fetch: requesting chunk 10
Aug 14 11:00:49 rasperry librespot[3490]: DEBUG:librespot::spirc: kMessageTypeNotify "raspotify (rasperry.com)" 2cb0f80d51724694a8409b54290eec7963edd616 3 1502708449684
....
Aug 14 11:00:49 rasperry pulseaudio[3549]: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Aug 14 11:00:49 rasperry pulseaudio[3549]: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Aug 14 11:00:49 rasperry pulseaudio[3549]: [pulseaudio] bluez4-util.c: org.bluez.Manager.GetProperties() failed: org.freedesktop.DBus.Error.UnknownMethod: Method "GetProperties" with signature "" on interface "org.bluez.Manager" doesn't $
Aug 14 11:00:49 rasperry bluetoothd[580]: Endpoint registered: sender=:1.56 path=/MediaEndpoint/A2DPSource
Aug 14 11:00:49 rasperry bluetoothd[580]: Endpoint registered: sender=:1.56 path=/MediaEndpoint/A2DPSink
....
Aug 14 11:00:52 rasperry librespot[3490]: DEBUG:librespot_audio::fetch: File 3dbcb2e7557c4ce1cca3893a6ba52656217dbab2$
Aug 14 11:01:06 rasperry librespot[3490]: DEBUG:librespot::spirc: kMessageTypeNext "GT-I9195I" 13b432bc1352df2454869a$
Aug 14 11:01:06 rasperry librespot[3490]: DEBUG:librespot::player: command=Load(SpotifyId(u128 { high: 39914449116091$
Aug 14 11:01:07 rasperry librespot[3490]: DEBUG:librespot::spirc: kMessageTypeNotify "raspotify (rasperry.com)" 2cb0f$
Aug 14 11:01:07 rasperry librespot[3490]: INFO:librespot::player: Loading track "Bad Kingdom"
Aug 14 11:01:07 rasperry librespot[3490]: DEBUG:librespot_audio::fetch: Downloading file 82a24e356db1a55099c3aaab0dc1$
....
Aug 14 11:01:07 rasperry librespot[3490]: INFO:librespot::player: Track "Bad Kingdom" loaded
....
Aug 14 11:01:10 rasperry librespot[3490]: DEBUG:librespot::spirc: kMessageTypeSeek "GT-I9195I" 13b432bc1352df2454869a$
[... cut ...]
Aug 14 11:02:17 rasperry librespot[3490]: DEBUG:librespot::spirc: kMessageTypeNotify "GT-I9195I" 13b432bc1352df2454869a740835594f211e6e7d 140 1502708506441
Aug 14 11:02:20 rasperry librespot[3490]: DEBUG:librespot::spirc: kMessageTypeHello "GT-I9195I" 13b432bc1352df2454869a740835594f211e6e7d 141 1502708506441
Aug 14 11:02:20 rasperry librespot[3490]: DEBUG:librespot::spirc: kMessageTypeNotify "raspotify (rasperry.com)" 2cb0f80d51724694a8409b54290eec7963edd616 18 1502708540760
Aug 14 11:02:47 rasperry librespot[3490]: DEBUG:librespot_core::session: Session[0] strong=3 weak=4
```

maciekbaczmanski, 2017-08-14:
Btw When trying to run it with librespot command:
`$ librespot --name "librespot test" --backend alsa --cache /tmp/librespot-test --bitrate 160 --verbose`
Everything works great

tomtana, 2017-08-14:
The command works only if you are running it as root or to with the user which is logged in to the desktop. I guess here lies the issue with pulse audio. When running as raspotify user it is conflicting with the desktop user.

tomtana, 2017-08-14:
I've found a fix if deinstalling pulseaudio is not an option for you. You have to make pulseaudio compatible for the use with multiple users. I followed this (http://billauer.co.il/blog/2014/01/pa-multiple-users/) tutorial and adapted it to raspotify.

Here is what you have to do:
1.change to your desktop user, in my case it's pi
`su pi
cd ~/`

2. copy the global config file to your user directory and change ownership
`mkdir ~/.pulse/
sudo cp /etc/pulse/default.pa ~/.pulse/
sudo chown -R pi:pi ~/.pulse`

3. go to the raspotify user home dir
`cd /var/cache/raspotify
sudo mkdir .pulse
sudo echo "default-server = 127.0.0.1" > .pulse/client.conf
sudo chown -R raspotify:raspotify ~/.pulse`

4. reboot and enjoy

maciekbaczmanski, 2017-08-14:
I get permission denied on 3rd step after command `sudo echo "default-server = 127.0.0.1" > .pulse/client.conf`

maciekbaczmanski, 2017-08-14:
But anyways I think You misunderstood me. I've uninstalled pulseaudio and it still doesnt work :(

maciekbaczmanski, 2017-08-14:
Maybe I didn't uninstall it completely? How can I check it?

dtcooper, 2017-08-14:
@tomtana it looks like you're exactly right. Pulse doesn't want to be shared between the user I create (`raspotify`) and the desktop user. This follows that it all works for me using Jessie Lite (since I have no desktop environment).

Everyone, I'll give it some thought and see if I can push a fix soon. Open to suggestions!

tomtana, 2017-08-14:
@maciekbaczmanski sorry there was an error in the commands, i didnt test them... there was also one step missing. I've updated them now. You can uninstall it with `sudo apt-get purge pulseaudio` . You could check it maybe with `dpkg --list | grep pulse` and see if there are still some packages installed

@dtcooper unfortunately I have very little knowledge of the linux audio stack to be of any help. As deep as I could research the problem I have 3 different solutions in mind. Unfortunately seems none of them to be clean clean enough to put into a package:

1. modify pulseaudio to accept tcp as explained here (http://billauer.co.il/blog/2014/01/pa-multiple-users/)
2. modify pulseaudio to run in system-mode as explained here (https://www.freedesktop.org/wiki/Software/PulseAudio/Documentation/User/SystemWide/)
3. run raspotify as the standart desktop user. (But what happens if multiple users want to use the desktop?)

maciekbaczmanski, 2017-08-14:
Tried everything... I've reinstalled my raspbian, installed only raspotify, tried playing with pulseaudio uninstalled (spotify thinks it plays, but nothing comes out and cant pause the song), tried Your code @tomtana - the same thing. I think problem lies somewhere else, but I don't know where... Btw librespot command still works everytime. I'm slowly starting to think about just using this command on startup on my desktop user (it's only for my smart room so it hopefully won't be overloaded or smthng)

tomtana, 2017-08-14:
@maciekbaczmanski did u do a restart after purging pulseaudio? However, you could change the user under which raspotify is executed in the file 
`sudo nano /etc/systemd/system/multi-user.target.wants/raspotify.service`
to your desktop user and run
`sudo systemctl daemon-reload`
and reboot

maciekbaczmanski, 2017-08-14:
WOW! After reinstalling my sys one more time and uninstalling pulseaudio before installing raspotify everything works correctly!! big thanks @dtcooper  adn @tomtana  for great support :D

closed 2017-08-14 by maciekbaczmanski

morph85, 2017-12-25:
Hi, confirming @tomtana's fix `sudo apt-get purge pulseaudio` rectified the issue where Spotify music cannot go beyond 3 or 5 tracks. Yay!

AndyBoey, 2018-01-24:
Hi, confirming @tomtana's comment on Aug 15 about making Raspotify work with PulseAudio, might be useful for those who needs to keep PulseAudio (e.g. running pi as an A2DP sink).

Works on my Pi 3

Bibo-Joshi, 2018-03-14:
@tomtana, @AndyBoey coud you specify, which `<user>` and `<group>` to use?
I'm running `raspotify` on `openhabian` ([openhabian](https://docs.openhab.org/installation/openhabian.html))  and tried with `openhab` and `sudo`, but that didn't work.

Until now it worked like a charm. Only recently, I encounter the "stops after a few songs" error. Maybe I messed something up, while setting up [`Jasper`](http://jasperproject.github.io/documentation/installation/#configure-raspbian)? I'm not sure, but I think the error didn't occur before …

I tried removing/purging `pulseaudio`, but then `raspotify` won't even start (as @maciekbaczmanski reported). Also, running `librespot --name "librespot test" --backend alsa --cache /tmp/librespot-test --bitrate 160 --verbose` gives the same error  :(

Since I've put quite some work into the Pi by now, I'm not really keen to recreate @maciekbaczmanski's fix of Aug. 14th (reinstalling system) …