dtcooper/raspotify #14: Install error on RPi3 running OSMC (Debian)

zilexa, 2017-08-24:
I tried your .sh automated install and the manual method. I am not a Linux expert at all, but this is what happened: 

Automated (last part): 
`Suggested packages:
  lsb
The following NEW packages will be installed:
  apt-transport-https lsb-release
0 upgraded, 2 newly installed, 0 to remove and 1 not upgraded.
Need to get 162 kB of archives.
After this operation, 269 kB of additional disk space will be used.
Get:1 http://ftp.debian.org/debian/ jessie/main apt-transport-https armhf 1.0.9.8.4 [135 kB]
Get:2 http://ftp.debian.org/debian/ jessie/main lsb-release all 4.1+Debian13+nmu1 [26.5 kB]
Fetched 162 kB in 0s (1,460 kB/s)
Selecting previously unselected package apt-transport-https.
(Reading database ... 32153 files and directories currently installed.)
Preparing to unpack .../apt-transport-https_1.0.9.8.4_armhf.deb ...
Unpacking apt-transport-https (1.0.9.8.4) ...
Selecting previously unselected package lsb-release.
Preparing to unpack .../lsb-release_4.1+Debian13+nmu1_all.deb ...
Unpacking lsb-release (4.1+Debian13+nmu1) ...
Setting up apt-transport-https (1.0.9.8.4) ...
Setting up lsb-release (4.1+Debian13+nmu1) ...
Raspotify installer only runs on a Raspberry Pi
`

Manual: 
`osmc@ChilleTV:~$ sudo apt-get -y install apt-transport-https
Reading package lists... Done
Building dependency tree
Reading state information... Done
apt-transport-https is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
osmc@ChilleTV:~$ curl -sSL https://dtcooper.github.io/raspotify/key.asc | sudo apt-key add -v -
gpg: pub  4096R/7953A455 2017-06-15  David Cooper <david@dtcooper.com>
OK
osmc@ChilleTV:~$ echo 'deb https://dtcooper.github.io/raspotify jessie main' | sudo tee /etc/apt/sources.list.d/raspotify.list
deb https://dtcooper.github.io/raspotify jessie main
osmc@ChilleTV:~$ sudo apt-get -y install raspotify
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package raspotify
`

I guess some checks are done to verify it is indeed a Raspbery Pi, somehow RPIs running OSMC (https://osmc.tv/) are not recognised properly?

dawidd6, 2017-08-30:
You need to perform `sudo apt update` after adding new repository

zilexa, 2017-08-31:
Gosh can't imagine I forgot to do that, installation was successful!

dtcooper, 2017-09-06:
What does `lsb_release -a` return on OSMC? @zilexa

dawidd6, 2017-09-06:
@dtcooper well, on my OSMC install there is no lsb_release at all, isn't that an ubuntu thing only?

zilexa, 2017-09-11:
Sorry for the delayed response, Raspotify works flawlessly so I didn't check back. 

```No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 8.9 (jessie)
Release:        8.9
Codename:       jessie
```

zilexa, 2017-09-18:
Just FYI, Raspotify works as expected on OSMC. But I did create a feature request on OSMC forum to create a Kodi addon for OSMC devices based on LibreSpot: 
https://discourse.osmc.tv/t/librespot-in-the-osmc-app-store/38717

Just to natively integrate it in OSMC/Kodi. 
LibreElec did the same thing: https://forum.libreelec.tv/thread/8438-librespot-addon/

OSMC is a great alternative to LibreElec since it is build on Debian and much more flexible.

HA4g3n, 2017-11-05:
Hello,

I get the same error: _E: Unable to locate package raspotify_
When i run: _sudo apt-get update_

i get the following:
_W: Failed to fetch https://dtcooper.github.io/raspotify/dists/jessie/InRelease  Unable to find expected entry 'main/binary-arm64/Packages' in Release file (Wrong sources.list entry or malformed file)

E: Some index files failed to download. They have been ignored, or old ones used instead._

I have reeboted and tried several commands that i can think of.
Im trying to install on my Vero 4K with OSMC installed over a SSH session.

Thanks
HA4g3n

JasonLG1979, 2021-11-16:
Closing because of age and inactivity. A new version of Raspotify based on a much newer version librespot will be released soon. When that happens please update and if the issue persists reopen the issue.

closed 2021-11-16 by JasonLG1979