dtcooper/raspotify #15: Piping this to another app

1liminal1, 2017-09-01:
Hi guys, 

Just wondering if you could help me, how can I run this as a named pipe to output it to forked-daapd. Im using it as an Airplay Server for Spotify.

chimpy, 2017-09-12:
Hi,

You should be able to add an option in the config of `--backend pipe`.

The pipe should be in the library being read by forked-daapd.

1liminal1, 2017-09-12:
Thats awesome!, ill try this tonight! 

Thanks

chimpy, 2017-09-12:
Take a look at the last couple of replies here: https://community.home-assistant.io/t/dacp-daapd-itunes-support/779

Make sure to remove the `--backend` argument from the systemd file: `etc/systemd/system/multi-user.target.wants/raspotify.service` otherwise it'll complain in the syslog about it being defined twice.

1liminal1, 2017-09-13:
Ok, followed what you wrote, but having issues, its not showing as a connect device.

Does this look right to you?

```
# /etc/default/raspotify -- Arguments for librespot

# Device name on Spotify Connect
#DEVICE_NAME="AirPlay"

# Bitrate, one of 96 (low quality), 160 (default quality), or 320 (high quality)
BITRATE="160"

# Additional command line arguments for librespot can be set below.
# See `librespot -h` for more info.
#
# To make your device visible on Spotify Connect across the Internet add your
# username and password which can be set via "Set device password", on your
# account settings, use `--username` and `--password`.
#
# To choose a different output device (ie a USB audio dongle or HDMI audio out),
# use `--device` with something like `--device hw:0,1`. Your mileage may vary.
#
#OPTIONS="--username "username" --password "password" --$  --backend pipe --device /home/pi/music/spotify"
```

1liminal1, 2017-09-13:
Ok, was working off some incorrect assumptions, more me and Linux then anything else. 

Ive gotten it to here; but I still cant see it as a connect device. 

```
pi@AirServer:~ $ sudo systemctl status raspotify -l
● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled)
   Active: active (running) since Wed 2017-09-13 16:03:28 AEST; 2s ago
  Process: 4403 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 4400 ExecStartPre=/bin/mkdir -p 0755 /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 4406 (librespot)
   CGroup: /system.slice/raspotify.service
           └─4406 /usr/bin/librespot --name AirPlay --bitrate 160 --disable-audio-cache --username username --password password --backend pipe --device /home/pi/music/spotify

Sep 13 16:03:28 AirServer systemd[1]: Stopping Raspotify...
Sep 13 16:03:28 AirServer systemd[1]: Starting Raspotify...
Sep 13 16:03:28 AirServer systemd[1]: Started Raspotify.
Sep 13 16:03:28 AirServer librespot[4406]: INFO:librespot: librespot ddfc28f (2017-08-04). Built on 2017-08-07. Build ID: N5RcEuis
Sep 13 16:03:29 AirServer librespot[4406]: INFO:librespot_core::session: Connecting to AP "gae2-accesspoint-b-tffj.ap.spotify.com:4070"
```

1liminal1, 2017-09-13:
Ok, now I have 

```
● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled)
   Active: activating (auto-restart) (Result: exit-code) since Wed 2017-09-13 16:18:38 AEST; 5s ago
  Process: 5311 ExecStart=/usr/bin/librespot --name ${DEVICE_NAME} --bitrate ${BITRATE} $CACHE_ARGS $OPTIONS (code=exited, status=101)
  Process: 5308 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 5305 ExecStartPre=/bin/mkdir -p 0755 /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 5311 (code=exited, status=101)

Sep 13 16:18:38 AirServer systemd[1]: raspotify.service: main process exited, code=exited, status=101/n/a
Sep 13 16:18:38 AirServer systemd[1]: Unit raspotify.service entered failed state.
```

chimpy, 2017-09-13:
@1liminal1 I'm pretty sure you just have a permissions problem that's all.

Go back to what you were doing in your penultimate post. It seems to me that the client had connected correctly, and you say it works when you do it directly.

You just need to _either_ correct permissions on `/home/pi/music/spotify` or change that path to somewhere that the `raspotify` user can read/write. I suggest the following:

`sudo mkdir -p /opt/music`
`sudo mkfifo /opt/music/spotify`
`sudo chown -R raspotify /opt/music`

Now change `--device /home/pi/music/spotify` to `--device /opt/music/spotify`

1liminal1, 2017-09-13:
Thanks Chimpy, followed it to the letter, but still the same. 

Im not sure if its a permissions issue, if I leave the defaults, it should still show up as a local spotify device right? Its not, and if I add my password ect, it seems to hang on 

```
17-08-04). Built on 2017-08-07. Build ID: N5RcEuis
Sep 14 09:36:52 AirServer librespot[4969]: INFO:librespot_core::session: Connecting to AP "gae2-accesspoint-b-v7w7.ap.spotify.com:4070"
```

Now, im pretty new to linux, so I could be wrong in what i'm assuming here. What do you think?

1liminal1, 2017-09-22:
Got this running 

--disable-discovery 

I think adding that fixed the issue :)

Thanks for all the help

Its very slow though. Taking 10-20 seconds to change songs/ volume. 

Anyone have any ideas on that?

chimpy, 2017-09-22:
Never had this problem.

1liminal1, 2017-10-05:
Yeah, I have seen that. 

Thanks :)

ricardovdheijden, 2018-03-07:
Hi, I seem to have the same issue as @1liminal1 in the beginning. However I cannot get it working.
My setup is: raspotify + forked-daap with library folder /srv/music. The pipe "spotify" is placed in this folder.
With the default configuration of raspotify I am able to play music through a connected headphone.

I gave the folowing permissions:
`sudo chmod -R 757 /srv/music/`
(yes it is overkill, but just being sure)

After that i noticed @chimpy 's note about permissions for raspotify:
`sudo chown -R raspotify /srv/music`

The only thing I am not confident about, is the config line for OPTIONS:
`OPTIONS="--username "username" --password "password" --$  --backend pipe --device /srv/music/spotify --disable-discovery"`
I tried it with and without --disable-discovery and I removed the # in front assuming this activates that line of configuration.

When I restart the raspotify service I see:
```
pi@raspberrypi:/srv/music $ sudo systemctl status raspotify.service -l
● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Wed 2018-03-07 20:47:47 UTC; 9s ago
  Process: 19670 ExecStart=/usr/bin/librespot --name ${DEVICE_NAME} --backend alsa --bitrate ${BITRATE} $CACHE_ARGS $OPTIONS (code=exited, status=101)
  Process: 19667 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 19664 ExecStartPre=/bin/mkdir -p 0755 /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 19670 (code=exited, status=101)

Mar 07 20:47:47 raspberrypi systemd[1]: raspotify.service: Main process exited, code=exited, status=101/n/a
Mar 07 20:47:47 raspberrypi systemd[1]: raspotify.service: Unit entered failed state.
Mar 07 20:47:47 raspberrypi systemd[1]: raspotify.service: Failed with result 'exit-code'.
```

Any idea what I am missing? Thanks in advance :)

dtcooper, 2018-03-24:
Dupe of #83, which has been closed. Thanks!

closed 2018-03-24 by dtcooper