dtcooper/raspotify #16: Not working on Orange PI

djsuszi, 2017-09-03:
It's very similar board, working too on Rasbian

Maybe you can check what is wrong.

./librespot -n a
INFO:librespot: librespot ddfc28f (2017-08-04). Built on 2017-08-07. Build ID: N5RcEuis
thread 'main' panicked at 'called Result::unwrap() on an Err value: Error { repr: Os { code: 92, message: "Protocol not available" } }', /checkout/src/libcore/result.rs:860
stack backtrace:
0: 
1: 
2: 
3: 
4: 
5: 
6: 
7: 
8: 
9: 
10: 
11: 
12: 
13: __libc_start_main

strace:

write(2, "INFO:librespot: librespot ddfc28"..., 87INFO:librespot: librespot ddfc28f (2017-08-04). Built on 2017-08-07. Build ID: N5RcEuis) = 87
write(2, "\n", 1
) = 1
socketpair(PF_LOCAL, SOCK_STREAM|SOCK_CLOEXEC|SOCK_NONBLOCK, 0, [6, 8]) = 0
getrandom(0xbefe5717, 0, GRND_NONBLOCK) = -1 ENOSYS (Function not implemented)
open("/dev/urandom", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 9
ioctl(9, FIOCLEX) = 0
read(9, "\36\266\200U\315\242\2\365", 8) = 8
read(9, "\345\323\f\255\232\22\317{", 8) = 8
close(9) = 0
rt_sigaction(SIGINT, {0xb6a4aaf0, [], SA_RESTORER|SA_RESTART|SA_SIGINFO|SA_NOCLDSTOP, 0xb6866271}, {SIG_DFL, [], 0}, 8) = 0
epoll_ctl(3, EPOLL_CTL_ADD, 6, {EPOLLIN|EPOLLOUT|EPOLLRDHUP|EPOLLET, {u32=2, u64=2}}) = 0
getrandom(0xbefe889f, 0, 0) = -1 ENOSYS (Function not implemented)
open("/dev/urandom", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 9
ioctl(9, FIOCLEX) = 0
read(9, ".\244\322D@q\200W}c5\355\242\365G\5\261\216!\336\217fx==\355\4x\267\375c!"..., 1024) = 1024
close(9) = 0
socket(PF_INET, SOCK_STREAM, IPPROTO_IP) = 9
ioctl(9, FIOCLEX) = 0
setsockopt(9, SOL_SOCKET, SO_REUSEADDR, [1], 4) = 0
bind(9, {sa_family=AF_INET, sin_port=htons(0), sin_addr=inet_addr("0.0.0.0")}, 16) = 0
listen(9, 1024) = 0
fcntl64(9, F_GETFL) = 0x2 (flags O_RDWR)
fcntl64(9, F_SETFL, O_RDWR|O_NONBLOCK) = 0
epoll_ctl(3, EPOLL_CTL_ADD, 9, {EPOLLIN|EPOLLOUT|EPOLLRDHUP|EPOLLET, {u32=4, u64=4}}) = 0
getsockname(9, {sa_family=AF_INET, sin_port=htons(60312), sin_addr=inet_addr("0.0.0.0")}, [16]) = 0
uname({sysname="Linux", nodename="orangepi2", ...}) = 0
socket(PF_INET, SOCK_DGRAM, IPPROTO_IP) = 10
ioctl(10, FIOCLEX) = 0
setsockopt(10, SOL_SOCKET, SO_REUSEADDR, [1], 4) = 0
setsockopt(10, SOL_SOCKET, SO_REUSEPORT, [1], 4) = -1 ENOPROTOOPT (Protocol not available)
close(10) = 0
socket(PF_INET6, SOCK_DGRAM, IPPROTO_IP) = 10
ioctl(10, FIOCLEX) = 0
setsockopt(10, SOL_SOCKET, SO_REUSEADDR, [1], 4) = 0
setsockopt(10, SOL_SOCKET, SO_REUSEPORT, [1], 4) = -1 ENOPROTOOPT (Protocol not available)
close(10) = 0
close(9) = 0
write(2, "thread '", 8thread ') = 8
write(2, "main", 4main) = 4
write(2, "' panicked at '", 15' panicked at ') = 15
write(2, "called Result::unwrap() on an "..., 111called Result::unwrap() on an Err value: Error { repr: Os { code: 92, message: "Protocol not available" } }) = 111
write(2, "', ", 3', ) = 3
write(2, "/checkout/src/libcore/result.rs", 31/checkout/src/libcore/result.rs) = 31

maybe you have some knowledge what is wrong ?

HausnerR, 2017-09-03:
add --disable-discovery for OPTIONS in config.

It's probably some problem with librespot and this workaround issue.

LonelyEagle29, 2020-12-31:
Update: I got working without issues on Orange Pi Lite, just installing, no type of further configuration needed.

JasonLG1979, 2021-11-16:
Closing because of age and inactivity. A new version of Raspotify based on a much newer version librespot will be released soon. When that happens please update and if the issue persists reopen the issue.

closed 2021-11-16 by JasonLG1979