dtcooper/raspotify #17: No control over looping if played on raspotify device

michalfita, 2017-09-03:
If the track/album is played from **Spotify** on the device running **raspotify** I don't have control over the looping options - pressing the loop icon doesn't switch the mode.

May be related to #8.

rokjsn, 2017-09-13:
Yeah, I really need this too

leojonathanoh, 2017-09-24:
Yeah, repeat and shuffle don't seem to work. Pressing those buttons visually enables them on Spotify App, but in actuality repeat and shuffle are not enabled. Would love for this feature to be included or fixed.

NeusAap, 2017-12-07:
Yup, would be awesome. Seems to still be the problem.

reaper, 2018-01-21:
Same, I realized too that shuffle and loop buttons aren't working on my android phone. Can you please bring us a fix asap ? By the way thanks for the port.

Elalith, 2021-01-12:
Does anyone else still have this issue? I noticed it a while back, but already saw this ticket. Unfortunately.. the issue still exists. It would make an already great application even better!

PS. I'm not complaining. The fact that this application even exists is one of the reasons I love open source software!

closed 2021-11-24 by JasonLG1979