dtcooper/raspotify #18: Access is Denied?

Taconut, 2017-09-08:
Hi! When I try to play a song over Spotify Connect, I get the following error:
`
ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Access denied
thread '<unnamed>' panicked at 'called `Option::unwrap()` on a `None` value', /checkout/src/libcore/option.rs:335
note: Run with `RUST_BACKTRACE=1` for a backtrace.
`

I've tried adding user raspotify to the audio group, but that didn't do anything. Any ideas?

Taconut, 2017-09-08:
Figured it out! All you have to do is add raspotify to the pulse-access group if running from PulseAudio. For the lazy, that's `sudo usermod -a -G pulse-access raspotify`. Everything works perfectly now. Thanks  for the amazing program!

closed 2017-09-08 by Taconut

dtcooper, 2017-09-08:
Awesome. I'll add this to the install script.

benhmin, 2021-12-21:
This still seems to be an issue. I have been trying to get raspotify to work all day from a clean install on bullseye. Was finally able to get it to work after reading this thread.

Followed steps to install raspotify the easy way. Would sometimes get an error at the end saying "usermod: user 'raspotify' does not exist"

was getting the same pulseaudio access denied error. Same error with librespot from command line but worked great when running librespot as sudo.

I had to first create the user "raspotify" and then add to the pulse-access group as per alexhulbert, now it's working.

JasonLG1979, 2021-12-21:
> This still seems to be an issue. I have been trying to get raspotify to work all day from a clean install on bullseye. Was finally able to get it to work after reading this thread.
>
> Followed steps [...]

Well I'm not sure what you did but that's not it if you installed Raspotify from a fresh install. The Raspotify service does not have anything to do with a raspotify user, or any user at all in fact. It uses the dynamic user systemd functionality.