dtcooper/raspotify #19: volume drops randomly

jb2210, 2017-09-09:
hi there,

I started using this application a litte while ago on a headless raspberry pi, and it worked perfectly till today. The volume keeps dropping randomly and I can't find the problem. I looked for the .log file but couldn't find it. Which makes it impossible to include in this issue.

I hope to see some response and thanks in advance for looking into it.

markheloking, 2017-10-13:
It's handy (even when you would have been able to find logs) to give the dev more info if you actually want this fixed. Questions the dev would probably like to have answers to are:

1. Does the volume change to 0 (effectively mute), or just decrease in volume to a lower volume level (eg 100% -> 50%)?
2. Does it happen at the start of a new track, or also during a track?
3. Does the volume only decrease, or does it also randomly increase?
4. Does the volume decrease multiple times consecutively? aka does it go down 100% -> 50% and then 50% -> 25% or something like that?
5. Are you able to increase the volume manually after it has decreased?
6. Is it a specific song/playlist/situation where this happens?
7. Are you able to do something specifically to fix it/prevent it from happening?

jb2210, 2017-12-09:
After some measuring I found out the problem wasn't much related to raspotify but rather to my HIFI system. At least that's what I think, I'm not really sure since I also changed to an external audio card (due to the bad audio quality of the default jack plug on the rpi). This could also have affected/resolved the problem. 

@noamelf I would suggest getting yourself a external usb audio card. If you already are using an external audio card; I'm in the dark of what your problem could be.

@markheloking the loudness just kept switching between, between let's say 80% and 20%. Not in a sequence or special order, just random. I will keep the thing about librespot in mind, good to know!

closed 2017-12-12 by jb2210