dtcooper/raspotify #20: Keeps crashing

1liminal1, 2017-09-14:
Hi guys, 

raspotify keeps crashing, anyone seen this before? Running from a Pi Zero W

```
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled)
   Active: activating (auto-restart) (Result: exit-code) since Thu 2017-09-14 13:28:04 AEST; 6s ago
  Process: 2952 ExecStart=/usr/bin/librespot --name ${DEVICE_NAME} --bitrate ${BITRATE} $CACHE_ARGS $OPTIONS (code=exited, status=101)
  Process: 2949 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 2946 ExecStartPre=/bin/mkdir -p 0755 /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 2952 (code=exited, status=101)

Sep 14 13:28:04 AirServer librespot[2952]: note: Run with `RUST_BACKTRACE=1` for a backtrace.
Sep 14 13:28:04 AirServer systemd[1]: raspotify.service: main process exited, code=exited, status=101/n/a
Sep 14 13:28:04 AirServer systemd[1]: Unit raspotify.service entered failed state.
```

derWeihnachtsmann, 2017-09-14:
Sadly, That's not enough debug info for helping..

1liminal1, 2017-09-14:
I'm pretty new to Linux, I do know about tail, and I've had a look in /var/logs but I cant find out which one it might be?

Do you know where the logs are?

Thanks

1liminal1, 2017-09-22:
Ok, I think it was  --disable-discovery that fixed this. 

Working now, just slow.

closed 2017-09-22 by 1liminal1

Zylvian, 2019-02-23:
> Ok, I think it was --disable-discovery that fixed this.
> 
> Working now, just slow.

Where did you do the command?