dtcooper/raspotify #21: Spotify not found with android version of Spotify

koenrot, 2017-09-18:
I just created a clean installation of Raspbian with the raspotify package. The raspberry is immediatly found with Spotify on my Windows desktop and works perfectly. Unfortunately it is not found by Spotify on an Android Smartphone. I tried two phones, both connected to the same network, which gave the same result. 

Only just after I start playing a song from my desktop, it pop ups on my phone. I have no issues with other Spotify Connect enabled audio players. 

Any idea?

zilexa, 2017-09-18:
I think this has to do with Spotify Connect itself, not with Raspotify or LibreSpot. The behavior you are descibing is almost identical to how Sonos works with Windows/Android: I do not see the Sonos device on Spotify Win32 application on Win10. Only if I start playing a song with my Android device on the Sonos, it pops up on Windows. I blamed Sonos for being a closed system, but it seems to be how Spotify Connect works. No issues if you only use 1 type of OS as source.

dtcooper, 2017-09-18:
Uhhhh, what @zilexa said. 😜 Closing.

closed 2017-09-18 by dtcooper

koenrot, 2017-09-19:
Too bad! Thanks for your reply.