dtcooper/raspotify #22: Spotify only playing back from HDMI

KaiiSt, 2017-09-24:
Hello there,I just went through the easy installation and setup of raspotify, no problems at all. I have my Raspberry Pi 3 connected to a TV via HDMI and the analog audio output connected to a hifi-system. The RPi runs OSMC and I tried all the audio settings that include the analog output (analog only and analog+hdmi). In the config of raspotify I used the command --device hw:0 to send audio to the analog output.

No matter what I choose, Spotify only sends sound to the hdmi out. That means I'd have to keep on the TV (which is wired to the hifi-system as well). Did I miss an option where I can specify the output of Raspotify?

Thanks a lot!

dtcooper, 2017-09-25:
I can try to help. Could you paste the contents of your `/etc/default/raspotify` please!

KaiiSt, 2017-09-26:
GNU nano 2.2.6          File: /etc/default/raspotify                        

/etc/default/raspotify -- Arguments for librespot

Device name on Spotify Connect
DEVICE_NAME="Raspberry"

Bitrate, one of 96 (low quality), 160 (default quality), or 320 (high quality)
BITRATE="320"

Additional command line arguments for librespot can be set below.
See `librespot -h` for more info.

To make your device visible on Spotify Connect across the Internet add your
username and password which can be set via "Set device password", on your
account settings, use `--username` and `--password`.

To choose a different output device (ie a USB audio dongle or HDMI audio out),
use `--device` with something like `--device hw:0,1`. Your mileage may vary.

OPTIONS="--username XXXXXX --password XXXXXXX --device hw:0"

----------------------------------------------------
This is it. I couldn't get the code to show up as a "whole codeblock". I got rid of the commentmarks since they turned everything in huge letters.

berrnd, 2017-10-01:
Hi,

had the same problem and stumbled over this while searching ... got it working by adding `--onstart 'sudo amixer cset numid=3 1' --onstop 'sudo amixer cset numid=3 0'` to `OPTIONS`.

See the [Raspberry Pi documention](https://www.raspberrypi.org/documentation/configuration/audio-config.md), you also maybe have to add `raspotify ALL = (root) NOPASSWD: /usr/bin/amixer` to your `/etc/sudoers`.

magnulu, 2017-10-01:
Thank you so much for this fix. OSMC first-timer, also raspotify virgin. You made my day :-)

mreigosa, 2017-11-08:
the same problem here, is this the solution?:
OPTIONS="--onstart 'sudo amixer cset numid=3 1' --onstop 'sudo amixer cset numid=3 0'"

WolfgangHusen, 2017-12-02:
Help folks, 
same issues on all my raspis: sound via HDMI works, but not on Jack.
Here my option file (nothing special - at the bottom you see some things I tried according the recommendations above):
# /etc/default/raspotify -- Arguments for librespot

# Device name on Spotify Connect
DEVICE_NAME="JajaSpot"

# Bitrate, one of 96 (low quality), 160 (default quality), or 320 (high quality)
#BITRATE="160"

# Additional command line arguments for librespot can be set below.
# See `librespot -h` for more info.
#
# To make your device visible on Spotify Connect across the Internet add your
# username and password which can be set via "Set device password", on your
# account settings, use `--username` and `--password`.
#
# To choose a different output device (ie a USB audio dongle or HDMI audio out),
# use `--device` with something like `--device hw:0,1`. Your mileage may vary.
#
#OPTIONS="--username <...> --password <pw>"

# Uncomment to use a cache for downloaded audio files. It's best to leave this
# as-is since permissions are properly set on directory `/var/cache/raspotify'.
#CACHE_ARGS="--cache /var/cache/raspotify"

#--onstart 'sudo amixer cset numid=3 1'
#--onstop 'sudo amixer cset numid=3 0'

#OPTIONS="--onstart 'sudo amixer cset numid=3 1' --onstop 'sudo amixer cset numid=3 0' --device 'hw:0,0'"
#OPTIONS="--onstart 'sudo amixer cset numid=3 1' --onstop 'sudo amixer cset numid=3 0'"
OPTIONS:"--device 'hw:0,0'"

STATUS here (seems to be fine):
pi@jajaraspi:~ $ sudo systemctl status raspotify -l
● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled)
   Active: active (running) since Sa 2017-12-02 14:46:46 CET; 4s ago
  Process: 24901 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 24898 ExecStartPre=/bin/mkdir -p 0755 /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 24904 (librespot)
   CGroup: /system.slice/raspotify.service
           └─24904 /usr/bin/librespot --name JajaSpot --backend alsa --bitrate 160 --disable-audio-cache

Dez 02 14:46:46 jajaraspi systemd[1]: Started Raspotify.
Dez 02 14:46:46 jajaraspi librespot[24904]: INFO:librespot: librespot 8971d3a (2017-10-05). Built on 2017-10-05. Build ID: POyxPQVg
Dez 02 14:46:47 jajaraspi librespot[24904]: WARN:mdns::fsm: error sending packet Error { repr: Os { code: 99, message: "Cannot assign requested address" } }

And on the Spotify Browser app I can select the raspi and it runs, but no sound.

jb2210, 2017-12-09:
@WolfgangHusen you could try using this thread to awnser your question:
https://www.raspberrypi.org/forums/viewtopic.php?f=26&t=11086

This is the awnser given:

> In that case you can force it from a command line prompt...
Code: Select all
`sudo amixer cset numid=3 <n>`
where n is 0=auto, 1=headphones, 2=hdmi. 
Change <n> to 1 to force sound through the 3.5mm headphone jack  :D 

Since I'm not sure this command is persistent after reboot, just follow the following steps to make the change persistent after boot.

To keep this change, just add this line to the crontab file:
if you are using a rpi with desktop, open a terminal, if not just issue the commands in the command line interface.
1. change to root user
`sudo su`
password is prompted, just enter your sudo password (user password)
2. open the crontab file
`crontab -e`
3. add the line (using the nano file editor)
add this line at the bottom of the file:
`@reboot amixer cset numid=3 <1>`
this runs the command after every boot
4. if using nano file editor, issue 'ctrl-x', then 'y', then just press enter.
reboot and see if it worked.

WolfgangHusen, 2017-12-21:
thanks @jb2210, with your feedback and the other comments I could make it work.

xtraminty, 2018-03-29:
I was having the same issue and just wanted to post the (fairly) simple fix I found for it. It's a little messy but it still works anyway.

Anyway, I'm running OSMC, which as it's not Raspbian means I can't use the Raspi config method or anything with the amixer. Instead what I found to do was edit the actual /boot/config.txt file using a simple sudo nano and added `hdmi_ignore_edid_audio=1`

I don't fully understand why it works but I think the command stops any sound to the display and so the service instead defaults to the analogue output. I presume this command would also mean anything else you had running that wasn't OSMC/kodi would also default to analogue audio. OSMC still works fine with audio coming out over HDMI and/or analogue so it only seems to be for

scred47, 2019-03-24:
> I was having the same issue and just wanted to post the (fairly) simple fix I found for it. It's a little messy but it still works anyway.
> 
> Anyway, I'm running OSMC, which as it's not Raspbian means I can't use the Raspi config method or anything with the amixer. Instead what I found to do was edit the actual /boot/config.txt file using a simple sudo nano and added `hdmi_ignore_edid_audio=1`
> 
> I don't fully understand why it works but I think the command stops any sound to the display and so the service instead defaults to the analogue output. I presume this command would also mean anything else you had running that wasn't OSMC/kodi would also default to analogue audio. OSMC still works fine with audio coming out over HDMI and/or analogue so no issues there

This is what worked for me same setup osmc with raspberrypi2

closed 2021-05-16 by KaiiSt

Delphae, 2021-10-30:
What worked for me was the suggestion of F2210:
run on the command line
sudo amixer cset numid=3 1

the audio is sent to the jack output
I am running Volumio on a Raspberry 3B
after reboot it remains sending the audio to jack :-)