dtcooper/raspotify #23: Raspbian with HiFiBerry - ALSA equalizer not working

3579km, 2017-09-24:
I have Raspberry Pi 2 with installed Raspbian Lite from HiFiBerry download page. It has been modified to automatically detect and configure of the HiFiBerry DAC+ Pro board. I installed Raspotify and it works great: spotify connect and sound, and then i installed ALSA equalizer from this manual https://www.hifiberry.com/build/documentation/guide-adding-equalization-using-alsaeq/

But editing equalizer to manual setting not working, sound not changes.
I tested it, play file by command: aplay /path/to/file.wav 
and i hear changes, edited in equalizer.

Maybe it's the problem with setting in Raspotify?

markheloking, 2017-10-04:
Directly from that link you sent:

> The equalization works only when the application uses the default output device from asound.conf. Applications that are configured to use the hardware device directly (you might see “hw:0” or similar in the command line options or in the configuration file) won’t use the equalization plugin.

Aka this plugin does not work for Raspotify since Raspotify uses the audio hardware directly instead of the default audio output device. I think that's actually part of librespot (which raspotify uses) and not raspotify itself.

schlachtkreuzer6, 2018-01-09:
Is it possible to change that in anyway? 
I found this, but I didn't try it.
http://www.gerrelt.nl/RaspberryPi/wordpress/equalizer/

RealZac, 2018-01-27:
> Is it possible to change that in anyway?
I found this, but I didn't try it.
http://www.gerrelt.nl/RaspberryPi/wordpress/equalizer/

I tried it, but it doesn't work. Raspotify still uses the audio hardware directly.

RealZac, 2018-07-22:
Hi,
is there anything new about this? Or maybe an other way to control bass and treble?
Thanks.

ger-xXfreshXx, 2018-09-01:
Hi,
I'm looking forward to a solution.
I'm not a hardware/ linux developer but I found in /etc/defaults/raspotify this information:

> To choose a different output device (ie a USB audio dongle or HDMI audio out),
> use `--device` with something like `--device hw:0,1`. Your mileage may vary.

Can this help us to get an equalizer for raspotify?

Thanks in advance!

FredRanG, 2018-10-24:
ALSA Equalizer working now with sudo -u raspotify alsamixer -D equal and OPTIONS="--device equal" in /etc/default/raspotify.

RealZac, 2018-11-05:
> ALSA Equalizer working now with sudo -u raspotify alsamixer -D equal and OPTIONS="--device equal" in /etc/default/raspotify.

Perfect! Thank you.

schlachtkreuzer6, 2018-11-24:
Great thx

agonzalezv, 2019-03-27:
> After adding parameter OPTIONS="--device equal" no sound. Any ideas why?

- `rm -rf ~/.asoundrc` (good idea to make a copy before)
- `sudo touch /etc/asound.conf`

- Then follow these instructions: https://www.hifiberry.com/build/documentation/guide-adding-equalization-using-alsaeq/ , make sure you add the contents of `asound.conf` as described. 

- `sudo vi /etc/default/raspotify`, add `OPTIONS="--device equal"`
- `alsactl kill rescan`
- `sudo systemctl daemon-reload`
- `sudo systemctl restart raspotify`
(or `sudo reboot now` to restart the system)
- To change EQ settings: `sudo -u raspotify alsamixer -D equal`

Sanderei, 2020-04-11:
Can't do it, it says that file or map doesn't exist

MarshMelone, 2020-07-05:
to whom it may concern:

I use 2 raspberry 2. For the first the ```"sudo -u raspotify alsamixer -D equal"``` works out of the box to set the equalizer with raspotify.

For the second it didn't work. I don't know why.
I had to edit ```"sudo nano /lib/systemd/system/raspotify.service"```.

I changed the line:
```"ExecStart=/usr/bin/librespot --name ${DEVICE_NAME} $BACKEND_ARGS --bitrate ${BITRATE} $CACHE_ARGS $VOLUME_ARGS $OPTIONS --device hw:0,0"```

to:
```"ExecStart=/usr/bin/librespot --name ${DEVICE_NAME} $BACKEND_ARGS --bitrate ${BITRATE} $CACHE_ARGS $VOLUME_ARGS $OPTIONS --device equal"```

I changed only the name of the device to the previously set device "equal" (see the link which was told in the comments above).
Now the mixer works with the second raspi too via: ```"sudo -u raspotify alsamixer -D equal"```

I hope i can lower the stress level of someone with this info :-)

Sanderei, 2020-08-05:
It doesn't work for me, raspotify can't access the equalizer, because the ¨operation is not permitted¨.

closed 2021-11-24 by github-actions[bot]