dtcooper/raspotify #29: Playback via external DAC

mankowskiwik, 2017-10-13:
Hi all, 

I am unable to play sound through external dac (Denon DA-300USB)
PreviouslyI was using fornoth solution for spotify connect and I was passing `-Dplughw:DA300USB `parameter for configuration. This does not work here. 

I also tried `--device hw:1`, which does not help. 

Device output: 
```

 0 [ALSA           ]: bcm2835 - bcm2835 ALSA
                      bcm2835 ALSA
 1 [DA300USB       ]: USB-Audio - DA-300USB
                      D & M Holdings Inc. DA-300USB at usb-3f980000.usb-1.2, high speed
```
```
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: DA300USB [DA-300USB], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

Could you please help me? Thanks a lot!

snizzleorg, 2017-10-31:
use ```--backend pipe --device <some fifo>``` and install forked-daapd. forked-daapd will play form the pipe to all your airplay speakers.

julianwiggins, 2020-11-23:
This worked a treat for me! —device hw:2,0 and I was away!

closed 2021-11-24 by JasonLG1979