dtcooper/raspotify #30: Any way to make it multiroom?

henrikklarup, 2017-10-16:
Is there any way to install it on multiple pis to make it play multiroom in sync?

markheloking, 2017-10-16:
You could set up multiple pi's running raspotify and then use [this](https://www.cnet.com/how-to/how-to-setup-multiroom-music-with-google-cast/) method to set it up as a multiroom system I'd presume, but I must say that I haven't tried it. I have multiple pi's laying around, so I might test it if I'm bored sometime.

henrikklarup, 2017-10-16:
I can not seem to find any of my pis with raspotify in the Google Home app, any other ideas?

derWeihnachtsmann, 2017-10-16:
U can use Snapcast(github) as combination with librespot. Works pretty good here.

henrikklarup, 2017-10-16:
Yes, but then it is not natively supported inside the Spotify client

derWeihnachtsmann, 2017-10-16:
yes it is, trough Librespot's own Discovery Feature. The Snapcast-Server just grabs the audio from librespot and sends it to its Snapcast-Clients.  👍

henrikklarup, 2017-10-16:
Then I've gotta try that

christianfl, 2017-10-30:
@derWeihnachtsmann Do you maybe have a short explanation of how to use raspotify with snapcast? I couldn't find any documentation for this specific combination - or is it just install-and-forget? Raspotify itself is working like a charm here.

wall0159, 2018-04-08:
I've got my raspberry Pi broadcasting to FM using a small USB to FM dongle. Then I just tune in the house's radios to that station and I have multi-room. Sound quality is very good.

derWeihnachtsmann, 2018-04-12:
Sorry for the very very late answer, just put `SNAPSERVER_OPTS="-d -s spotify:///librespot?name=snapcast&bitrate=160" in your /etc/default/snapserver file. 
`

5AMsan, 2018-04-23:
Hey found this issue searching a way to use Raspotify across several RPIs. already configured to use a master Mopidy. I changed Raspotify configuration file from :
`#BACKEND_ARGS="--backend alsa"`
to :
`BACKEND_ARGS="--backend pipe --device /tmp/snapfifo"`
allowing to use both Mopidy and Raspotify on my RPI network.

bitttttten, 2018-08-29:
How did you get Snapcast running on the pi?

dmckminard, 2018-10-16:
@5AMsan , can you explain a bit more what have you done ? I can't make it working
If I try to put this : 

> SNAPSERVER_OPTS="-d -s spotify:///librespot?name=snapcast&bitrate=160" in your /etc/default/snapserver file

I have no sound when Mopidy is working, only when Raspotify is working.
And if I use your command, I don't have any Spotify sound, and it bugs (sometimes disappears in my Spotify devices when I'm connected)

dmckminard, 2018-10-16:
> @dmckminard <https://github.com/dmckminard> , I assume you didn't setup mopidy to output audio through snapcast pipe but you did in librespot using SNAPSERVER_OPTS config.
> I did not change SnapServer config :
>
> # defaults file for snapserver
>
> # start snapserver automatically?
> START_SNAPSERVER=true
>
> # Allowed options:
> #   -h, --help                          Produce help message
> #   -v, --version                       Show version number
> #   -p, --port arg (=1704)              Server port
> #   --controlPort arg (=1705)           Remote control port
> #   -s, --stream arg (=pipe:///tmp/snapfifo?name=default)
> #                                       URI of the PCM input stream.
> #                                       Format: TYPE://host/path?name=NAME
> #                                       [&codec=CODEC]
> #                                       [&sampleformat=SAMPLEFORMAT]
> #   --sampleformat arg (=48000:16:2)    Default sample format
> #   -c, --codec arg (=flac)             Default transport codec
> #                                       (flac|ogg|pcm)[:options]
> #                                       Type codec:? to get codec specific options
> #   --streamBuffer arg (=20)            Default stream read buffer [ms]
> #   -b, --buffer arg (=1000)            Buffer [ms]
> #   --sendToMuted                       Send audio to muted clients
> #   -d, --daemon [=arg(=0)]             Daemonize
> #                                       optional process priority [-20..19]
> #   --user arg                          the user[:group] to run snapserver as when daemonized
>
> USER_OPTS="--user snapserver:snapserver"
>
> #SNAPSERVER_OPTS="-s pipe:///tmp/snapfifo"
> But I configured both Mopidy and Raspotify to output through SnapServer pipe.
> My (relevant) Mopidy conf (/etc/mopidy/mopidy.conf):
>
> [audio]
> output = filesink location=/tmp/snapfifo
> And the relevant Raspotify conf (/etc/default/raspotify):
>
> BACKEND_ARGS="--backend pipe --device /tmp/snapfifo"

Yes thanks you a lot, it works !

(Je peux peut-être te parler en français, ça serait plus simple si ça te convient!)
Il y a néanmoins un petit soucis avec Raspotify. La musique se lit en accélérée, j’entends par là qu’elle va plus vite que la normale, as-tu été confronté également à ce problème ?

Merci beaucoup en tout cas!

5AMsan, 2018-10-16:
I'll keep it english for GitHub readers visiting from search engine if you don't mind.
Never had pitch issues on Raspotify stream but I read about slower speed while playing on some devices :/
#129

dmckminard, 2018-10-16:
No problems ! Yes, I've already seen that, I'm going to search and post if I found something!

mikkelrask, 2018-10-17:
Snapcast is a seperate server, which simply plays on all running snapcast clients, correct? 

What if I would like to group my raspotify server with my Chromecast Audio in my kitchen, for example? Any ways of making raspotify, librespot or similar show up in my Google Home application, so I could do that? 

I got a spare Pi so i _could_ go for a dual pi setup, along the snapcast option, but i'd like to use the chromecast in my kitchen for stuff like soundcloud, radio apps etc.

Hydro8, 2018-11-29:
Hi
Is it possible to send multiple stream (multiple raspotify ?) to snapcast ?
I want to be able to listen my spotify music through connect and my wife too through her own account.

Hydro8, 2018-11-29:
So if inside the config file of raspotify I specify account, player will only be avaible from spotify connect for this account ?

Then, can I set 2 instance of raspotify for each player to have 2 streams inside snapcast ? (maybe I don't fully understand how it works)

jshep321, 2019-11-16:
Hey all, thanks for the magic line:
BACKEND_ARGS="--backend pipe --device /tmp/fifofilenamegoeshere"

Even though this is closed, I'm adding some details that might potentially help others!

I have connected **raspotify to forked-daapd** for a multiroom (wifi/ airplay compatible) solution.  I like raspotify (spotify connect) better than the pseudo-integrated spotify client in forked-daapd.  Here is how it works:

1. create a fifo handler _in a directory that is automatically indexed by forked-daapd_ with 'mkfifo spotifyfifo'.  
2. Load raspotify and add the magic line above with /home/<user>/spotifyfifo to /etc/default/raspotify.  Restart raspotify (sudo systemctl restart raspotify).
3. load spotify, point to your server, and it magically takes over forked-daapd to play.

(Note that you can create a forked-daapd airplay "repeater" in the same way with shairport-sync.)

As a side note, playing spotify connect fifo and airplay sink at the same time creates an unhappy race condition.

hessius, 2020-05-08:
@jshep321 I tried this as user pi on a raspbian setup, but can’t get it to work.
Would the device be /home/pi/spotifyfifo ?

When I try the raspotify ”speaker” on Spotify I just get stuck on ”connecting”

jshep321, 2020-05-14:
Well the device needs to be in your "music" directory from your forked-daapd.conf file.  The default setting is pasted below.  

`        # Directories to index
       directories = { "/srv/music" }
`

Unless your music files, remote files, etc. are in /home/pi/ then it won't index them.  Or you can add /home/pi to the conf file, but that is a bit ugly (using your homedirectory that is).

febs, 2020-10-02:
> I've got my raspberry Pi broadcasting to FM using a small USB to FM dongle. Then I just tune in the house's radios to that station and I have multi-room. Sound quality is very good.

Is this legal? 🤔

jshep321, 2020-10-03:
> 
> 
> > I've got my raspberry Pi broadcasting to FM using a small USB to FM dongle. Then I just tune in the house's radios to that station and I have multi-room. Sound quality is very good.
> 
> Is this legal? 🤔

In the US, probably.  Assuming you stay off of 87.7-87.9 and power+antenna gain is limited (measured in field strength), then yes.  
[https://recnet.com/unlicensed](url)

With a cheap USB dongle, you're likely OK, but I am not a lawyer. :)

wall0159, 2020-10-13:
@febs, I think it's ok - it's very low power. It's just receivable in the back and front yards, but barely from the street. I don't think even my next door neighbours would notice any interference.

G-Christ, 2020-11-08:
> 1. mkfifo spotifyfifo

Hey @jshep321, 

I am trying to follow your guide to use raspotify together with forked-daapd
Here is a part of my  /etc/forked-daapd.conf
##############################################
# Library configuration
library {
        # Name of the library as displayed by the clients (%h: hostname). If you
        # change the name after pairing with Remote you may have to re-pair.
        name = "My Music on %h"

        # TCP port to listen on. Default port is 3689 (daap)
        port = 3689

        # Password for the library. Optional.
#       password = ""

        # Directories to index
        directories = { "/srv" }
##############################################################

I created fifo handler spotifyfifo in  directory /svr

I added
BACKEND_ARGS="--backend pipe --device /srv/spotifyfifo"  
to /etc/default/raspotify

restareted both  services  raspotify and forked-daapd

From Spotify I point at raspotify  

In forked-daapd I can see spotifyfifo as a source, but cannot play it.

I am afraid that raspotify  is not streaming the output to spotifyfifo.
Can you see anything wring ion my approach?
If not, do you have any idea for how I can check what went wrong?

Thanks, 
Ghassan

jshep321, 2020-11-08:
Hi,
Everything looks correct to me.  Maybe this is a permissions issue?
Also, make sure the existing BACKEND_ARGS line is commented out or deleted.
```
#BACKEND_ARGS="--backend alsa"
BACKEND_ARGS="--backend pipe --device /home/pi/music/spotifyconnect"
```
I have my music directory inside the user home directory (/home/pi/music), which is where my raspotify fifo is (as you can see above.

You can either try to replicate this in your home directory and/or make sure that the permissions on these are writable by all (or change ownership with chown).
```sudo chmod ugo+rw /srv/spotifyfifo``` and then restart.

G-Christ, 2020-11-08:
> Hi,
> Everything looks correct to me. Maybe this is a permissions issue?
> Also, make sure the existing BACKEND_ARGS line is commented out or deleted.
>
> ```
> #BACKEND_ARGS="--backend alsa" [...]

Hi @jshep321,

I replicated your setup but it still doesn’t work. But now I noticed a new error in my log:

#############################################################################
pi@mediaserver:/ $ tail -f /var/log/forked-daapd.log
[2020-11-08 21:57:47] [  LOG]     main: mDNS init
[2020-11-08 21:57:47] [  LOG]     mdns: Avahi state change: Client running
[2020-11-08 21:57:47] [  LOG]       db: Configured to use database file '/var/cache/forked-daapd/songs3.db'
[2020-11-08 21:57:47] [  LOG]       db: Now vacuuming database, this may take some time...
[2020-11-08 21:57:47] [  LOG]       db: Database OK with 3 active files and 6 active playlists
[2020-11-08 21:57:47] [  LOG]   laudio: Pulseaudio failed with error: Connection refused
[2020-11-08 21:57:47] [  LOG]   laudio: Error initializing Pulseaudio: Connection refused
[2020-11-08 21:57:48] [  LOG]     scan: Bulk library scan completed in 1 sec
#############################################################################

I tried top start/Stop the Pulseaudio service but I got this error, indicated that it is not installed at all?

#############################################################################
pi@mediaserver:/ $ sudo systemctl start pulseaudio
Failed to start pulseaudio.service: Unit pulseaudio.service not found.
#############################################################################

Do I need to install it? If so what fir and how do I do that?

Thanks  :)

jshep321, 2020-11-09:
Make sure your audio { section in /etc/forked-daapd.conf has no local pulse enabled.  Mine looks like this:
```
# Type of the output (alsa, pulseaudio, dummy or disabled)
#       type = "alsa"
```

G-Christ, 2020-11-09:
> Make sure your audio { section in /etc/forked-daapd.conf has no local pulse enabled. Mine looks like this:
> 
> ```
> # Type of the output (alsa, pulseaudio, dummy or disabled)
> #       type = "alsa"
> ```

Mine is already like this, and I think the issue is in the input stream and not the output.
can I some how verify that the output stream from raspotify is genereted correctly 

any other ideas?

G-Christ, 2020-11-09:
> Make sure your audio { section in /etc/forked-daapd.conf has no local pulse enabled. Mine looks like this:
> 
> ```
> # Type of the output (alsa, pulseaudio, dummy or disabled)
> #       type = "alsa"
> ```

One more thing
I noticed in the log: "No Spotify refresh token found" 

Do I need such a token? 

I thought it not necessary in this setup as daapd just streams  pipe no matter whether it is generated from Spotify or something else.
right?

If I do need the Spotify token anyway, how do I get it?

G-Christ, 2020-11-11:
> Make sure your audio { section in /etc/forked-daapd.conf has no local pulse enabled. Mine looks like this:
> 
> ```
> # Type of the output (alsa, pulseaudio, dummy or disabled)
> #       type = "alsa"
> ```

Hi @jshep321 
just wanted to tell you that I just managed to get it working now 👍 
I made various things amongst which:
I moved my library files to a new location ( home/pi/mymusic) 
made sure all folder names are low case 
installed Pulseaudio 
and the last thing I did was to regrant ugo+rw privileges to spotifyfifo

Next step is to try to enable streaming to Bluetooth speakers

Thanks for your help :)

jshep321, 2020-11-11:
congrats!

closed 2021-11-17 by JasonLG1979