dtcooper/raspotify #31: Shows Up; But No Sound

benbartosik1, 2017-10-19:
I ran the install and everything is showing up in Spotify Connect but there's no sound going through. I have the pi set up as a Roon Bridge and it works fine for that.

harrybaumann, 2017-10-22:
Maybe you don't use the right audio output device. I managed to get my non-default USB audio device working by doing the following:

List the audio devices on your raspberry.
`aplay -l`
Look for the identifyer numbers of your device. In my case, my USB device is card 1, device 0.
The HDMI device for example is on card 0, device 1.

Open the service configuration file of raspotify.
`sudo nano /lib/systemd/system/raspotify.service`

Add a parameter to the ExecStart command: `--device hw:1,0` (where the first number is the card and the second is the device).

Save, close, restart service.
`sudo systemctl daemon-reload`
`sudo systemctl restart raspotify`

Done.

benbartosik1, 2017-10-22:
Thanks,

This may be a bit of basic question - but when I open all that up in Terminal, I don’t know how to save the config files. 
I can type something but don’t know how to save or run a command afterwards.

Thanks

@benbartosik

harrybaumann, 2017-10-22:
After editing the service configuration in nano, press `Ctrl+x` to save and close the file.

chang69, 2018-02-04:
@harrybaumann Thank you for providing help. Your solution solved my issue of no audio.

closed 2018-02-23 by dtcooper

Hammer8, 2018-04-20:
Hi, I tried the above but still no sound (Spotify sees Raspotify fine and says it is connected)...my Pi3 is connected to a Pro-Ject S2 DAC.  I am running Raspotify as part of DietPi.  And with DietPi, both Roon RAAT and HQPlayer NAA works fine.  Any thoughts?  Thank you!  Hammer8

deanfourie1, 2020-09-24:
I found that the wrong device was being used: 0,1.
I ran

aplay -l

card 1: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 3/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3

Then tested each device manually until I found the appropriate device.

speaker-test -c2 -twav -l7 -D plughw:0,1

You should hear a read out of your audio channels.

Then added it on raspotify config

OPTIONS="--device hw:1,0"

Worked for me.

amitkeret, 2021-01-02:
> Add a parameter to the ExecStart command: `--device hw:1,0` (where the first number is the card and the second is the device).

I realise this is a rather old thread... I'm posting this addition in case someone else runs into the same problem.

I was trying to get raspotify to work on RasPi Zero W which is streaming to a bluetooth speaker, which I've connected using `alsa` and a PCM definition in `/etc/asound.conf`.

Spotify was connecting to raspotify but I could not get any audio. In my case, though, `$ aplay -l` shows an empty list (not sure why).
The solution was to set the device to the bluetooth speaker as is was defined in `/etc/asound.conf`

```
/etc/asound.conf
pcm.btspeaker {
  type plug
  slave.pcm {
...

/etc/default/raspotify
OPTIONS="--device btspeaker"
```

Callenberg, 2021-01-02:
`$ aplay -l` shows only the hardware gear, `$ aplay -L` shows PCM, so btspeaker should be listed there.

I got Raspotify (Librespot) to work well with Bluetooth speaker by just adding the asound.conf without any fiddling of the Raspotify conf file.

I think you might have missed to define btspeaker as a controller, see the ctl.btspeaker part below. I also set the btspeaker as default which makes things easier. The conf file below also works with mpd given that btspeaker is added to its conf file.

Here is the part of the asound.conf that seems to work for me: 

````
defaults.bluealsa.delay 10000
  
pcm.btspeaker {
    type plug
    slave.pcm {
        type bluealsa
        device 'FC:58:FA:ED:57:60'
        profile 'a2dp'
    }
    hint {
        show on
        description 'Bluetooth speaker, FC:58:FA:ED:57:60'
    }
}
ctl.btspeaker {
	   type bluealsa
}
pcm.!default {
	type plug
	slave.pcm "btspeaker"
	hint {
        	show on
        	description 'btspeaker'
    }
}
ctl.!default {
	type bluealsa
}`

digixmax, 2021-03-21:
I currently set the OPTION parameter --device hw:X,0 in /etc/default/raspotify to hardcode output playing on hw device X, otherwise there is no sound from raspotify.

Rather than hardcoding the selection, how do I set up so that raspotify would play out on whichever device the system is currently uses as output device?

wizzy2breezy, 2022-06-12:
In my case, none of the above worked, not sure how much has changed since this was originally posted
Here's what I did:

1) I ran "librespot --device ?" to get the list of available audio devices.
I needed the HDMI audio output so I looked for:
[image]

Only thing needed was the device ID which is "hdmi:CARD=vc4hdmi0,DEV=0"
At least as of 06/2022, there's a conf file in /etc/raspotify/
(I had to change the permissions first before I could access the folder and file)

In the conf file, I added the line "LIBRESPOT_DEVICE="hdmi:CARD=vc4hdmi0,DEV=0" and that solved it.
Obviously in your case it can be a different audio device.

Instead of uncommenting the #LIBRESPOT_DEVICE="default" line, I simply added the
LIBRESPOT_DEVICE="hdmi:CARD=vc4hdmi0,DEV=0" below it
[image]

swedeback, 2022-08-15:
Thanks @wizzy2breezy! Your solution made it work for me.

leskaman, 2022-09-21:
@wizzy2breezy 
I was struggling for a while but your solution worked in a minute !!! Thank you so much 😊

Doublejay, 2024-03-15:
Worth nothing that your client might need a restart too....mine did 🙄