dtcooper/raspotify #32: Shuffle not working on Spotify Connect

mickvdv, 2017-10-25:
The shuffle function doesn't work when connected to the raspberry pi (raspotify).

Version: 0.9.1~librespot.20171005T125424Z.8971d3a

johan-andersson01, 2017-11-17:
I have the same experience on RBP3 running OSMC.

NeusAap, 2017-12-07:
I am having the same problem. Most recent version on RBP2. Anyone found a workaround?

jb2210, 2017-12-09:
same problem here, raspbian lite and desktop. kernel version 4.9
rpi 1 v2

johan-andersson01, 2017-12-09:
To be fair though, "Shuffle Play" seems to works fine. But that might not be available on the PC clients of Spotify.

MrJoki007, 2018-01-17:
Same problem

lbligh, 2018-01-25:
I also have this issue, I can toggle shuffle on from my client (either my iPhone or Macbook), but as soon as the next track starts (next on the playlist, does not shuffle at all), the button toggles off again, leading me to believe that this Spotify Connect client just doesn't support shuffle play? More information would be helpful! Anyone?

cortegedusage, 2018-01-25:
the original librespot didn't support shuffle, and repeat for that matter.

There is a new repo
https://github.com/librespot-org/librespot
which seems to be the new librespot home
this one has shuffle.

@dtcooper also herrernst has volume normalization in his repo, which is a
very welcome
adition, especially when listening to playlists.

lbligh, 2018-01-25:
Thank you @cortegedusage that sounds perfect! Are there any disadvantages in comparison with Raspotify? Aside from the shuffle issues I've been loving it.

cortegedusage, 2018-01-25:
as raspotify uses the same code there are no down sides, besides that it
offers no way
to install it as simple as raspotify.

lbligh, 2018-01-25:
Ok thanks very much @cortegedusage

rkok98, 2018-01-25:
I'm sorry but I'm new to this. Could you provide me the steps to install Librespot-org?

cortegedusage, 2018-01-25:
The beauty of raspotify is that it provides a clean and simple way to install and configure librespot.
The code in Librespot-org is just the code it needs to be compiled, something that isn't something I can explain to you simple. Simply because I can't compile myself...
That said, leaves you the following options.
- Learn how to compile Rust (although very useful, I don't think you can learn this overnight) :)

- Wait until dtcooper patches his branch of librespot with the shuffle patch, probably he will.

- Install Volumio2 on your raspberry pi and install the volspotconnect2 plugin from balbuze, to be 
   found on github, this would also give you volume normalization.

rkok98, 2018-01-25:
Thank you! I may try to learn compiling with Rust if I find some free time, otherwise I'll wait for dtCooper.

ComlOnline, 2018-01-29:
Just a heads up the guide on the [wiki](https://github.com/librespot-org/librespot/wiki/Compiling)
has all you need:

For rpi from start to finish
````
sudo apt update
sudo apt upgrade
sudo apt install build-essential portaudio19-dev curl git
curl https://sh.rustup.rs -sSf | sh
git clone https://github.com/librespot-org/librespot.git
```
reboot here (I've found it helps)
```
cd librespot
cargo build --release
```
and then leave it for a good while. 
All going well you will have librespot in ./target/release.

mpmartinek, 2018-01-29:
Is that supposed to build the new librespot together with the raspotify?

dtcooper, 2018-01-29:
@ComlOnline + @mpmartinek, I've updated Raspotify to use the [librespot-org fork](https://github.com/librespot-org/librespot). I'll push the build within a few minutes, you can update Raspotify using `apt-get` in your Pi and it should work. Thanks all!

closed 2018-01-29 by dtcooper

mpmartinek, 2018-01-29:
Thanks so much @dtcooper! I'll try it asap

Bibo-Joshi, 2018-01-29:
Hey there!
I've upgraded this morning and found, shuffle und repeat (context mode, track mode is _not_ available) to work like a charm. Many thanks at @dtcooper for your work :)

**Additional information for openHAB users**
I installed raspotify on the same Pi, I'm running [openHAB](http://www.openhab.org) on, to use it e.g. as alarmclock. In combination with [this python script](https://github.com/pmpkk/openhab_spotify-webconnect-api), I'm able to controll Spotify using the [Spotify Web API](https://developer.spotify.com/web-api/) via openHAB.
After adding shuffle/repeat-functionality to the python script, I found setting shuffle/repeat mode only works, if you toggle it _after_ you have started a playlist.
Leaving this here for people who might be aiming for a similar setup with openHAB ;)

johan-andersson01, 2018-01-29:
After running apt update and apt full-upgrade on OSMC, I can confirm the following features to work:

- Queuing
- Shuffle 

Many thanks @dtcooper !