dtcooper/raspotify #33: Sound volume

Tott3, 2017-10-26:
Is there any way to adjust the default sound volume? Whenever I start a song the volume is set to max.

wrskalle, 2017-11-03:
Did you get anywhere with this?

Tott3, 2017-11-03:
I did not, still waiting for an answer :/

markheloking, 2017-11-03:
As raspotify is basically just a wrapper around librespot, this is more a librespot limitation/quirk. The problem is that the developer of librespot actually anounced a while back that he quit the development of librespot altogether. See https://github.com/dtcooper/raspotify/issues/27.

ghosty89, 2017-12-06:
Librespot is declared as Unmaintened so I forked it (https://github.com/ghosty89/librespot) and modified it to set initial volume 50%. I also add a parameter --initial-volume that allows user to pass the initial volume. After this I forked raspotify (https://github.com/ghosty89/raspotify if you want to see the changes) and edited to build my librespot instead of plietar'sa librespot and build the .deb with make. Now in the raspotify config in /etc/default/raspotify  I can pass the parameter "--initial-volume"
e.g.
`OPTIONS="--username  USER --password PASS --initial-volume 20"` where 20 is the % of the volume

tedbarth, 2017-12-23:
May it be possible to read out the phones / Spotifies musik volume and use it as start value?

fossedihelm, 2017-12-29:
It was my first option but I've read [here](https://github.com/plietar/librespot/issues/150#issuecomment-284461310) that it is not possible. I've not investigated more about it. I just trusted him.

kloknibor, 2017-12-30:
I can set the volume from my iphone in the spotify app, but only with the buttons! To set the initial volume I’ve used alsamixer just run it and you can easily change the volume!

optima-software, 2018-01-14:
If you adjust alsamixer  volume, you reduce the maximum volume to that value. That's not what's  needed.

dtcooper, 2018-05-18:
Try @optima-software's solution and if it doesn't work let me know. Closing.

closed 2018-05-18 by dtcooper

febs, 2020-10-01:
This is actually now implemented and configurable in /etc/default/raspotify. The file is well commented and self explanatory.

Writing for those that may stumbe here before than the documentation as I did.