dtcooper/raspotify #34: password crypto

aycgit, 2017-10-30:
would it be possible to encrypt the spotify password so people shouldersurfing or people finding a way into your raspberry can't easily just see your password in plaintext?

low priority but useful

dtcooper, 2017-11-17:
Kind of, sort of, not really. You can login in your browser at spotify.com,

Profile > Account > Set Device Password. Then you'll have a numeric username with a obfuscated, long, weird password.

Hope that helps!

closed 2017-11-17 by dtcooper

Wqrld, 2017-11-17:
i dont see a "device password" option anywhere

dtcooper, 2017-11-17:
Hmmm, maybe it's only available to accts linked with Facebook. This would
be a librespot issue, IHMO

Wqrld, 2017-11-17:
since librespot requires it as a command parameter(?) wouldnt it be possible to add a seperate field for a basic encrypted password that is decrypted my raspotify itself?