dtcooper/raspotify #40: Audio out via USB DAC

audiogeek123, 2017-11-05:
Hello, I have the following problem:
I connected my active speakers via USB to my Pi 3. They are set as default and work perfectly in browser etc. and when I start librespot in the shell with this command:

`librespot --name raspberry --bitrate 320 --backend alsa --device hw:1`

so I thougt when I change the /etc/default/raspotify to this it would have the same effect:

...
`OPTIONS="--name raspberry --bitrate 320 --backend alsa --device hw:1"`
...

Unfortunately this doesn't work, do you have any hint for me?

Thank you very much! :-)

Greets

audiogeek123, 2017-11-06:
I also tried  what @harrybaumann in #31 suggested, but this also didn't work:

> Open the service configuration file of raspotify.
> sudo nano /lib/systemd/system/raspotify.service
> 
> Add a parameter to the ExecStart command: --device hw:1,0 (where the first number is the card and the second is the device).
> 
> Save, close, restart service.
> sudo systemctl daemon-reload
> sudo systemctl restart raspotify

dtcooper, 2017-11-08:
0. What error messages come up when you do `$ sudo systemctl status rastpofify`?

1. Does the `raspotify` user have access to `hw:1`, ie:

```
$ sudo -u raspotify librespot --name raspberry --bitrate 320 --backend alsa --device hw:1
```

2. The `bitrate`, `name`, and `backend` are already provided, so don't include them in `OPTIONS`. You should just have `--device hw:1`.

hjolli, 2017-11-08:
I had the same problem found an easy fix if you don't need the on board audio.
Basically disable the on board audio.
And set you usb dac/soundcard as the first audio card in alsa.

Disables/blacklists the onboard audio.
`sudo echo blacklist snd_bcm2835 >  /etc/modprobe.d/alsa-blacklist.conf`
Sets you usb/dac card as the default device.
`sudo echo options snd-usb-audio index=0 > /etc/modporbe.d/alsa-base.conf `

the just restart your pi and you are done.

worked for me with the original dragonfly usb dac.

hjolli, 2017-11-08:
I had the same problem found an easy fix if you don't need the on board audio.
Basically disable the on board audio.
And set your usb dac/soundcard as the first audio card in alsa.

Disables/blacklists the onboard audio.
`sudo echo blacklist snd_bcm2835 >  /etc/modprobe.d/alsa-blacklist.conf`
Sets you usb/dac card as the default device.
`sudo echo options snd-usb-audio index=0 > /etc/modporbe.d/alsa-base.conf `

the just restart your pi and you are done.

worked for me with the original dragonfly usb dac.

audiogeek123, 2017-11-08:
hello, thank you for your reply!
I deleted everything from `options` except  `--device hw:1`

This is what I get with `$ sudo systemctl status raspotify -l`:
```

● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled)
   Active: active (running) since Wed 2017-11-08 20:20:32 UTC; 58s ago
  Process: 1836 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 1833 ExecStartPre=/bin/mkdir -p 0755 /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 1839 (librespot)
   CGroup: /system.slice/raspotify.service
           └─1839 /usr/bin/librespot --name Spotify Raspi --backend alsa --bitrate 320 --disable-audio-cache --device hw:1

Nov 08 20:21:01 raspberrypi librespot[1839]: INFO:librespot_core::session: Authenticated as "XXXXXXX" !
Nov 08 20:21:01 raspberrypi librespot[1839]: INFO:librespot::audio_backend::alsa: Using alsa sink
Nov 08 20:21:01 raspberrypi librespot[1839]: INFO:librespot_core::session: Country: "DE"
Nov 08 20:21:02 raspberrypi librespot[1839]: INFO:librespot::player: Loading track "Flight Of The Cosmic Hippo"
Nov 08 20:21:07 raspberrypi librespot[1839]: INFO:librespot::player: Track "Flight Of The Cosmic Hippo" loaded
Nov 08 20:21:08 raspberrypi pulseaudio[1848]: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Nov 08 20:21:08 raspberrypi pulseaudio[1848]: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Nov 08 20:21:08 raspberrypi pulseaudio[1848]: [pulseaudio] bluez4-util.c: org.bluez.Manager.GetProperties() failed: org.freedesktop.DBus.Error.UnknownMethod: Method "GetProperties" with signature "" on interface "org.bluez.Manager" doesn't exist
Nov 08 20:21:08 raspberrypi librespot[1839]: thread '<unnamed>' panicked at 'called `Option::unwrap()` on a `None` value', /checkout/src/libcore/option.rs:335:20
Nov 08 20:21:08 raspberrypi librespot[1839]: note: Run with `RUST_BACKTRACE=1` for a backtrace.

```

But now I discovered that after I updated & upgraded everything i can delete all the additional `options` and it works!
But I still get some errors:

```
 ● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled)
   Active: active (running) since Wed 2017-11-08 20:26:44 UTC; 18min ago
  Process: 1940 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 1937 ExecStartPre=/bin/mkdir -p 0755 /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 1943 (librespot)
   CGroup: /system.slice/raspotify.service
           ├─1943 /usr/bin/librespot --name Spotify Raspi --backend alsa --bitrate 320 --disable-audio-cache
           └─2248 /usr/bin/pulseaudio --start --log-target=syslog

Nov 08 20:29:53 raspberrypi librespot[1943]: INFO:librespot::player: Track "Flight Of The Cosmic Hippo" loaded
Nov 08 20:44:20 raspberrypi librespot[1943]: INFO:librespot::player: Loading track "Flight Of The Cosmic Hippo"
Nov 08 20:44:31 raspberrypi librespot[1943]: INFO:librespot::player: Track "Flight Of The Cosmic Hippo" loaded
Nov 08 20:44:31 raspberrypi pulseaudio[2248]: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Nov 08 20:44:31 raspberrypi pulseaudio[2248]: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Nov 08 20:44:31 raspberrypi pulseaudio[2248]: [pulseaudio] bluez4-util.c: org.bluez.Manager.GetProperties() failed: org.freedesktop.DBus.Error.UnknownMethod: Method "GetProperties" with signature "" on interface "org.bluez.Manager" doesn't exist
Nov 08 20:44:31 raspberrypi pulseaudio[2248]: [alsa-sink-USB Audio] alsa-sink.c: ALSA woke us up to write new data to the device, but there was actually nothing to write!
Nov 08 20:44:31 raspberrypi pulseaudio[2248]: [alsa-sink-USB Audio] alsa-sink.c: Most likely this is a bug in the ALSA driver 'snd_usb_audio'. Please report this issue to the ALSA developers.
Nov 08 20:44:31 raspberrypi pulseaudio[2248]: [alsa-sink-USB Audio] alsa-sink.c: We were woken up with POLLOUT set -- however a subsequent snd_pcm_avail() returned 0 or another value < min_avail.
Nov 08 20:44:32 raspberrypi librespot[1943]: ALSA lib pcm.c:7843:(snd_pcm_recover) underrun occurred
```
What should I do about that?

@hjolli Wouldn't it be a problem if I disabled the internal soundcard and switch my speakers/DAC off and on again?

Thank you!

dtcooper, 2017-11-08:
@audiogeek123 if it works, I wouldn't worry about the errors. Think of them as warnings. Closing!

closed 2017-11-08 by dtcooper

mmarcm, 2017-11-20:
I have the same problem with my USB Bose Speakers.  The funny thing is it already worked a few months ago but I had to reinstall my RPI. 

```
$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: Audio [Bose USB Audio], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

OPTIONS is set to "--device hw:1,0" no further values (e.g. username, password)

```
$ sudo systemctl status raspotify -l
● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2017-11-20 14:40:19 CET; 7min ago
  Process: 6196 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/
  Process: 6193 ExecStartPre=/bin/mkdir -p 0755 /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 6199 (librespot)
   CGroup: /system.slice/raspotify.service
           └─6199 /usr/bin/librespot --name BOSE CONNECT --backend alsa --bitrate 320 --disable-audio-ca

Nov 20 14:40:19 raspberrypi systemd[1]: Starting Raspotify...
Nov 20 14:40:19 raspberrypi systemd[1]: Started Raspotify.
Nov 20 14:40:19 raspberrypi librespot[6199]: INFO:librespot: librespot 8971d3a (2017-10-05).
```

```
$ sudo -u raspotify librespot --name raspberry --bitrate 320 --backend alsa --device hw:1INFO:librespot: librespot 8971d3a (2017-10-05). Built on 2017-10-05. Build ID: POyxPQVg
INFO:librespot_core::session: Connecting to AP "lon6-accesspoint-a6.ap.spotify.com:4070"
INFO:librespot_core::session: Authenticated as "XXXX" !
INFO:librespot::audio_backend::alsa: Using alsa sink
INFO:librespot_core::session: Country: "DE"
INFO:librespot::player: Loading track "Little Talks"
INFO:librespot::player: Track "Little Talks" loaded
ALSA lib pcm.c:8389:(snd_pcm_set_params) Channels count (2) not available for PLAYBACK: Invalid argument
thread '<unnamed>' panicked at 'called `Option::unwrap()` on a `None` value', /checkout/src/libcore/option.rs:335:20
note: Run with `RUST_BACKTRACE=1` for a backtrace.
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: "SendError(..)"', /checkout/src/libcore/result.rs:860:4
```

I tried everything above but nothing is working.

audiogeek123, 2017-11-22:
It worked for me only a few days too. Here is how I solved it (until now):

`sudo nano /etc/modprobe.d/alsa-blacklist.conf`

add `blacklist snd_bcm2835`

save & close
reboot

`sudo nano /etc/modprobe.d/alsa-base.conf`

add `options snd-usb-audio index=0`

save & close
reboot

You should see now only your Bose USB Audio DAC when you type in `aplay -l`

Hope this works for you.

mmarcm, 2017-11-22:
It finally worked! Thank you for your solution! 
I had to remove the `--device hw:X,X` from OPTIONS even I selected the right output after blacklisting the internal sound card. (my Bose was now Card 0, device 0).

Thanks again for your help!