dtcooper/raspotify #41: Not Discoverable on local wifi - Pi 3

SimonSchaeferFinReach, 2017-11-11:
Hi,
Im using a rasperry pi 3. 
Unfortunately the pi is not discoverable on my local wifi right out of the box by any device.

I can only discover it once I modify the config file and add my username and password to it. 
This however means that everybody else on my network cannot use raspotify. 

Could you let me know how I can access the logs so I can give further information to you? 

Thanks,
Simon

t0k4rt, 2017-11-12:
Hi, 
Raspotify can only be used if you setup it with your credentials. it's not an Airplay player, it's a spotify connect client which means: 
* You can remote control songs played by raspotify through your spotify apps
* Raspotify uses your spotify premium account to directly stream content from spotify server 

Your raspotify setup will only be accessible to  an application that is connected to your premium account. The raspotify server won't be discoverable by any other device connected to your wifi network.

Hope it helps.

SimonSchaeferFinReach, 2017-11-20:
Hi,
Thanks for the reply. Thought so, but then maybe the readme should be changed, because it says that the service is discoverable on the local wifi, which made me think that its discoverable in spotify too. 

Still I would like to be able to allow multiple people using the Pi. Would it be possible to run multiple daemons on the pi and thus provide multiple login data?

Simon

dtcooper, 2017-11-20:
It's discoverable in Spotify _iff_ you have Spotify Premium. Do you? I'm happy to update the docs @SimonSchaeferFinReach.

SimonSchaeferFinReach, 2017-11-21:
I do have Spotify Premium, but its only discoverable once I input my login data in the config file. Which kinda contradicts the "it works right out of the box" statement in the docs, it should mention that you have to alter the config if you want it to work. 

Also do you think it would be possible somehow to setup raspotify for multiple people? For example by running multiple raspotify services?

Etuldan, 2017-12-23:
Same a SimonSchaeferFinReach, I thought raspotify was working out of the box for Premium users.
For exemple, i wanna install a raspi on my sound speaker.
Anyone in my wifi will be able to play any music, as soon as he is premium.

Also, for family accounts, it's quite useless, as only one account can be registered on raspotify.

Anyway, still a great and working app for single users :)

dtcooper, 2017-12-24:
@Etuldan, it _does_ work out of the box for Premium users. I confirm it working with the following,

1. Flash Raspbian Stretch Lite image on fresh Pi 3
2. Plug the Pi into router
3. Install using the one liner in the README
4. Log onto wifi network using Macbook
5. Start Spotify with Premium acct on Mac
6. See the device there "raspotify (raspberrypi)"
7. Play to it!

Closing. If you have more problems lemme know!

closed 2017-12-24 by dtcooper

Etuldan, 2017-12-24:
@dtcooper 
In my case : 
1. Stretch lite on Pi (B)
2. Plugin into router (ethernet)
3. One liner install
4. Wifi network using Win10 and/or Android
5. Start Spotify premium on Win10/Android
6. No raspotify displayed.

As soon as I set account credentials in the raspotify configuration, the 2 wifi devices see the rPI.

UPnP enabled on router.
May be related to some Apple autodiscovering protocol ?

I'm not reopening, as it seems related to someting external to raspotify ;)

radeusgd, 2018-06-06:
This may be obvious, but if it doesn't work out of the box, you may want to double-check your firewall settings on the PC and raspberry, because that may block discovery.