dtcooper/raspotify #45: Metadata

MrYom17, 2017-11-23:
Hi,

How could I read out metadata from raspotify ? (artist name etc)
I'd like display them on a hd44780

thanks

silberjan, 2017-11-25:
Similar request here: 
I want to call a webhook (to slack) everytime the song changes. It should post the song and artist to our #nowplaying slack channel.

Any idea how to extract this kind of information?

dtcooper, 2017-11-25:
@silberjan + @MrYom17: Have you tried polling [Spotify's Web API](https://developer.spotify.com/web-api/)?

silberjan, 2017-11-25:
i would call the `audio-analysis` endpoint if there is a way to get the currently playing songs id from raspotify (librespot)

dtcooper, 2017-12-27:
@MrYom17 you can either try the Web API or follow the instructions I laid out in #55 and use one of the forks mentioned there to track metadata changes. Thanks!

closed 2017-12-27 by dtcooper