dtcooper/raspotify #46: [Feature Request] Adjust sound volume via HDMI-CEC

christianfl, 2017-11-25:
Hi there!

Is it somehow possible to implement an option to adjust the Spotify sound volume over HDMI-CEC?
I am using an AV-receiver and the Raspberry Pi over HDMI is very quiet compared to other inputs. 

The receiver-integrated Spotify connection works like that: The volume bar within the Spotify app is relative so there is no total silent and no max volume. If you want it louder, it sets the receiver's volume higher. Unfortunately it is exposed to the internet so I prefer Raspotify over that integrated thing.

Maybe there is a possibility to do a similar thing?

marienfeldtom, 2017-11-28:
Raspotify uses librespot. You have to find a way to integrate volume control to librespot.

I already setup CEC on my raspberry (just for turning on/off the receiver).

Check out https://github.com/Pulse-Eight/libcec

dtcooper, 2017-12-27:
Please try @marienfeldtom's suggestion. Hopefully this works. Either way, HDMI volume control is outside of the scope of this project so closing.

closed 2017-12-27 by dtcooper