dtcooper/raspotify #47: Raspbian install issue

TheBrauxois, 2017-11-26:
During installation on Raspbian, I have this issue : 
`N: Skipping acquire of configured file 'non-free/binary-armhf/Packages' as repository 'http://repository.spotify.com stable InRelease' doesn't support architecture 'armhf'`

So the installation ends, I can see `raspotify (raspberry)` in my device list on all my devices, but can't connect to it..

dtcooper, 2017-11-27:
I'm having trouble understanding why http://repository.spotify.com was added to your sources... what command(s) did you use to install Raspotify?

dtcooper, 2017-11-27:
You could also try following the "(Hard Installation)[https://github.com/dtcooper/raspotify#hard-installation]" steps to manually install.

dtcooper, 2017-12-27:
Didn't receive feedback from @TheBrauxois so closing. It's my belief that the Spotify APT repository was adding to their config independent of anything to do with this software. If there are further related problems, please comment on this ticket and we'll re-open it.

closed 2017-12-27 by dtcooper