dtcooper/raspotify #48: --onstart parameter

marienfeldtom, 2017-11-27:
I've made a shell script, that turns on my receiver over CEC and want to have that script running with the --onstart parameter from librespot.

I changed the line of etc/default/raspotify to

OPTIONS="--onstart cd ../../ && sh home/pi/start.sh"

But if i start raspotify with this configuration it doesn't work and i can't even connect to my device.
What am I doing wrong?

fossedihelm, 2017-12-07:
try using 
`OPTIONS="--onstart /home/pi/start.sh"`
N:B. It will started when you play a song and not when you connect to the device.

dtcooper, 2017-12-27:
What @fossedihelm said! Hope that helps. You can also research google how to configure Raspotify's Systemd configuration to run a script before the daemon starts. Closing for now. If you need more help, please comment on this ticket and we'll re-open. Thanks!

closed 2017-12-27 by dtcooper

giggywithit, 2021-04-20:
> try using
> `OPTIONS="--onstart /home/pi/start.sh"`
> N:B. It will started when you play a song and not when you connect to the device.

Is there any way to get a short delay before playback using raspotify?
I have an amp that I have turned on when a track is selected, but it needs about a second to startup.
The track starts playing before the amp is ready and I would like to get a short pause here for this.