dtcooper/raspotify #49: Install error: Raspotify installer only runs on a Raspberry Pi

mikeapk, 2017-11-28:
Tried installing using the curl -sL.... command on a Raspberry Pi 2B that has OSMC installed. 

Got the error that: Raspotify installer only runs on a Raspberry Pi

Have tried both the easy and the hard/manual install options but get errors on both of them. 

Any thoughts?

mikeapk, 2017-11-28:
Got it working!

galorin, 2017-12-06:
https://xkcd.com/979/

riker65, 2017-12-14:
How did you solve it?

dtcooper, 2017-12-27:
I'm debating allowing Raspotify to install on OMSC (but undocumented, so people don't complain when _their_ particular setup breaks). I'll leave this ticket open, leave it pending as a feature request, and update you folks when I get a change to test it on a spare Pi I have laying around. It'll have to wait to the new year. Thanks!

jonsafc, 2018-01-08:
Hi, I'm quite a novice with the RPi3 which i bought the other week. I've installed OSMC via noobs and I'm trying to install Raspotify but I'm getting the message as the user above....... 'Raspotify installer only runs on a Raspberry Pi'. 

Am I right that your saying that it won't work on OSMC but will through Raspbian? Or other OS's?

Thanks

Gio2k, 2018-02-01:
Could you be more specific as to what breaks? I executed the instlalation steps manually, and apparently have no issues with the installation, but it seems the raspotify service is not running when i restart it

MrJoki007, 2018-02-02:
When you download the installer script (curl -sL https://dtcooper.github.io/raspotify/install.sh | sh)
just remove the exit 1 after echo "Raspotify installer only runs on a Raspberry Pi" that it looks like this and it will install on osmc
`run_on_pi_only() {
    echo "Raspotify installer only runs on a Raspberry Pi"
}`

dtcooper, 2018-02-03:
Hi @MrJoki007, could you please execute the command `lsb_release -si` on OSMC? Thank you!

reinoudk, 2018-02-16:
Would it be an option to change the check from `if lsb_release -si | fgrep -ivq raspbian; then` to `if uname -a | grep -Eivq 'armv7|armv6'; then`? This should ensure that the device architecture is at least compatible with the build from cargo. 

I've tested this on a Pi3 with OSMC, on which it works. Haven't had time yet to test it on a regular Raspbian image.

xumber, 2018-06-20:
I got it working on OSMC with the "hard" version install instruction:
super easy like 3 command in terminal
[](https://github.com/dtcooper/raspotify)

gfro84, 2018-07-15:
I'd suggest the stock answer to 'it doesn't work with my usb DAC speakers' is to blacklist the hdmi sound module thus making the usb the default. Trying to select the output device was a chore until I found this brutal but effective method. 

I'll try find the guide I followed if requested.