dtcooper/raspotify #50: help with editing raspotify options file

mikeapk, 2017-11-28:
Trying to get the output to go to my USB interface.

In the Raspotify file, do I need to remove the '#' characters? Add a device designator?

Trying to figure out exactly what I need to do in editing this file to steer the audio to USB.

Thoughts?

marienfeldtom, 2017-11-28:
You can find out your devices with `aplay -l  `

And yes, you have to remove # because it starts a comment.

For example my audio jack is hw:1

`OPTIONS="--device hw:1"`

dtcooper, 2017-12-27:
Yes, you'll need to remove the `#` beforehand unfortunately. The configuration files in `/etc/default` are typically environment variable files, ie. the equivalent of `source /etc/default/raspotify` is run before the daemon is kicked off using these environment variables. Lines with the hash (`#`) in shell are skipped, and lines without are read/run. Hope that helps! Thanks. Closing.

closed 2017-12-27 by dtcooper