dtcooper/raspotify #51: Multiple accounts with Spotify Connect

benirose, 2017-12-01:
Sadly, Spotify Connect is limited to a single account, but with Spotify Family, there's a need to be able to control a Spotify Connect device from different accounts within the same Family. Would it be possible to support multiple accounts with this client, maybe through multiple instances of librespot? Trying to find an alternative to using Airplay.

dtcooper, 2017-12-27:
I suppose something like multiple instances of librespot could fix the problem. You could try doing this by (a) installing raspotify and disabling the systemd service or (b) compiling librespot on your own -- and then creating your own individual services (I like [supervisord](http://supervisord.org/) for simple stuff, as its syntax is a little easier).

You could also try running Raspotify without specifying a `--username` or `--password` argument in its config file so everyone on your local intranet can discover and play music. YMMV with that, however.

Unfortunately, I won't be adding this as a use case for Raspotify, so I'm closing this ticket. In any event, thanks @benirose and happy hacking!

closed 2017-12-27 by dtcooper

benirose, 2017-12-28:
Awesome thanks for the detailed response! I’ll let you know what I discover if I ever get around to it!