dtcooper/raspotify #55: Get info is playing!

vjoxyodo, 2017-12-09:
Hello dtcooper,

Great lib with worked like charm. My question is there anyway of getting info of what is playing in raspotify? My intention to pull that info to a 16x2 LCD using Adafruit_CharLCD python lib. Any idea how to make this happen with your lib?

Cheers,
vjoxyodo

cortegedusage, 2017-12-09:
the librespot lib used in raspotify is currently unmaintained.
no one really stepped up to maintain it, there is however someone who forked it.
https://github.com/frafall/librespot

he has implemented output for metadata, maybe. dtcooper could use this fork.

swilalaa, 2017-12-10:
Would be happy to get the track & metadata into a pipe so forked-daapd can read it.

dtcooper, 2017-12-27:
Hey all! It's pretty easy to use someone else's  librespot fork that provides track info and build a debian package for your Pi. First, install [Docker](https://www.docker.com/), git (duh!), and Make (Make comes with the `build-essential` package on Debian/Ubuntu or on the Mac developer tools package)

Then clone Raspotify and enter the repo
```
git clone https://github.com/dtcooper/raspotify.git
cd raspotify
```

Then, pick your librespot fork and clone it, in this case user `example`,
```
git clone https://github.com/example/librespot
```

And run Make,

```
make
```

After a time, a package should be right there in your directory.

Anyway, since this arms you all with the stuff you need to get the current track, I'm closing this ticket. If you still have problems, point me toward a librespot fork you'd like me to build for you and I can compile it when I have time. Thanks!

closed 2017-12-27 by dtcooper

vorbiss, 2018-12-10:
Hi
is there another way to catch metadata to show them on screen ?

jshep321, 2019-11-19:
Hi @swilalaa 
Since you are ~a year in, are you happy with your (forked lib) metadata solution? 
Just got my raspotify -> forked-daapd setup running and I don't want to tear it apart for metadata unless yours still works well.   

Setup tips are welcome. Thanks!

Ulrar, 2020-06-05:
Hi @jshep321,
Did you end up doing it ? Looks like everyone ends up looking for a way to do this, there's a lot of issues about it on various repos.

jshep321, 2020-06-06:
> 
> 
> Hi @jshep321,
> Did you end up doing it ? Looks like everyone ends up looking for a way to do this, there's a lot of issues about it on various repos.

Nope, still on my todo list.

floodwayprintco, 2020-09-29:
@Ulrar @jshep321 Just chiming in here to say I am looking for a metadata solution as well.

Ulrar, 2020-09-30:
Hey,

I actually did end up doing it in my hassio addon : https://github.com/Ulrar/hassio-addons/tree/master/raspotify

I've had to update the patch myself to get the metadata pipe in there, feel free to use the patch or even the Dockerfile if you want !

floodwayprintco, 2020-09-30:
Thanks. Raspotify+Forked-Daapd on Hassio was my original attempt. It
skipped constantly and I'm pretty confident it's not a network issue based
on past experiences and messages in the log about not being able to process
the data stream fast enough along with everything else.

I just moved it to a standalone pi and it's working smoothly - everything
except the Metadata which was a nice feature of hassio but I'm not sure how
to update my current setup.

Can you give some more insight on the patch? Because I bet I could modify
those instructions for my setup and share that as well.

Ulrar, 2020-09-30:
The patch isn't mine, I've just updated it to get it to compile with the
current master branch but I may be able to help if you have specific
questions. I know just enough rust to keep these kinds of things compiling
I guess :)