dtcooper/raspotify #56: Volume always at max when connecting

tedbarth, 2017-12-10:
Is there a way to set the startup volume for raspotify that is applied when connection has started?

I use the newest raspotify successfully on my Raspberry Pi Zero Wifi with newest Raspbian. The problem I face is that when switching to it as device via Spotify app on my Android the attached device starts playing at 100% volume. This is so damn loud that even my boxes don't do more than clicking.

The 100% volume setting does not apply when connecting to other Spotify devices or when playing on the mobile phone.

fossedihelm, 2017-12-11:
Duplicate [33](https://github.com/dtcooper/raspotify/issues/33)

dtcooper, 2017-12-27:
I'm going to leave things at 100% for now, as this is a limitation of `librespot`. Please check out [@fossedihelm's fork of raspotify](https://github.com/fossedihelm/raspotify) and try installing his `.deb` file in his master branch.

If you'd like to keep using my project, you'll have to wait if/when I decide to pull in the `--initial-volume` change.

Thanks! Closing.

closed 2017-12-27 by dtcooper