dtcooper/raspotify #57: raspotify title / name does not change

danielsteiner, 2017-12-11:
Hi, I tried to change the name of raspotify in the config but even after restarting (and saving, and yes, I've verified that the file was saved) my raspberry pi, the name of the device is still displayed as raspotify (hostname of my pi) in spotify. 
Any ideas why this is?

fossedihelm, 2017-12-11:
Did you delete the `#` before DEVICE_NAME var? 
e.g.
`# Device name on Spotify Connect
DEVICE_NAME="raspotify-this-works-trust-me"
`

danielsteiner, 2017-12-12:
God I feel dumb now. That's what you get when you just rush trough the config. I didn't expect raspotify to simply work without changing _any_ configuration (since bitrate and options were commented out as well). 

thanks & can be closed

closed 2017-12-12 by danielsteiner