dtcooper/raspotify #58: no sound on hifiberry

riker65, 2017-12-14:
Hi
I am using the hifiberry aynd do not have any sound.

spotify so does recognize the device , starts, the song, but no sound.

I used the device option 
--device hifiberry   and also tried it without a special device - so default.

any resolution? Thanks 

my asound.conf is working for 
mpd and ario.

pcm.hifiberry { 
  type hw card 0 
}

pcm.!default { 
  type plug 
  slave.pcm "dmixer" 
}

pcm.dmixer { 
  type dmix 
  ipc_key 1024 
  slave { 
    pcm "hifiberry" 
    channels 2 
  } 
}

ctl.dmixer { 
  type hw 
  card 0 
}

riker65, 2017-12-15:
Update
with option -v I found error in log:
this might help.

INFO:librespot::player: Track "For The Love" loaded
thread '<unnamed>' panicked at 'called `Option::unwrap()` on a `None` value', /checkout/src/libcore/option.rs:335:20
note: Run with `RUST_BACKTRACE=1` for a backtrace.
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: "SendError(..)"', /checkout/src/libcore/result.rs:860:4

NCChris, 2017-12-17:
I have a hifiberry amp+ and had the same issue. You have to specify the audio output in the config file. First see if your hifiberry shows up: aplay -l
If you see it then go into the raspotify config file and add a config option to specify the audio to that device: sudo nano /etc/default/raspotify
add --device hw:0  (mine was 0, yours may be different) to the OPTIONS

dtcooper, 2017-12-27:
Hi @riker65, please try @NCChris's instructions. Closing for now, and feel free to re-open if you need help after trying that. Thanks!

closed 2017-12-27 by dtcooper

riker65, 2017-12-28:
Hi 
please reopen.
issue still is present.
even with @NCChris's solution.

I tried:
device hw:0
and  : --device dmix:CARD=sndrpihifiberry,DEV=0

which was provided by aplay -L

NF3RN0, 2019-07-04:
I had to blacklist the raspberry pi audio driver for it to work. I did not have to edit the OPTIONS in /etc/default/raspotify.

First:

`sudo nano /etc/modprobe.d/raspi-blacklist.conf`

add this line to raspi-blacklist.conf :

`blacklist snd_bcm2835`

Then:

`sudo reboot`

BUBUsnet, 2020-02-08:
My setup:
Raspberry Pi3 with OSMC and Hifiberry Digi+

I had this problem:
`Feb 08 10:29:07 pi3 librespot[1682]: ERROR:librespot_playback::audio_backend::alsa: Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'ENOENT: No such file or directory'
Feb 08 10:29:07 pi3 librespot[1682]: ERROR:librespot_playback::player: Could not start audio: Alsa error: PCM open failed
`

Solution:
I changed the device config to this and it worked: 
`--device hw:sndrpihifiberry hw:0`

To only have "hw:0" would not be sufficient. You can get the name of your soundcard with: aplay -l
`**** List of PLAYBACK Hardware Devices ****
card 1: sndrpihifiberry [snd_rpi_hifiberry_digi], device 0: HifiBerry Digi HiFi wm8804-spdif-0 []
  Subdevices: 0/1
  Subdevice #0: subdevice #0
`