dtcooper/raspotify #59: can not control master speaker with raspofity

riker65, 2017-12-15:
Hi
it it possible to control master volume.

alsamixer and amixer is showing mixer Speaker having USB device on Raspberry.

but using this mixer in raspotif is invalid? why?

Thanks for checking:
executing: 
/usr/bin/librespot --name OSMC95Wohn --backend alsa --bitrate 320 --disable-audio-cache --device default --mixer Speaker

leads to
thread 'main' panicked at 'Invalid mixer', /checkout/src/libcore/option.rs:819:4

but mixer is valid......

~$ amixer
Simple mixer control 'Speaker',0
  Capabilities: pvolume pswitch pswitch-joined
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 30
  Mono:
  Front Left: Playback 30 [100%] [0.00dB] [on]
  Front Right: Playback 30 [100%] [0.00dB] [on]

fossedihelm, 2017-12-15:
Maybe [this](https://github.com/plietar/librespot/issues/140) can help you

dtcooper, 2017-12-27:
Looks like @fossedihelm has you covered here. You'll want to use a little lower level abstraction of  "device" on ALSA. If you've got more problems, re-open. Closing.

closed 2017-12-27 by dtcooper