dtcooper/raspotify #60: cannot play songs: error sending packet

morals100, 2017-12-24:
Hello

When I try to play music through raspotify from my phone, it doesn't play. My phone says its playing but I hear no sound. I checked the logs with `sudo systemctl status raspotify` and the message `Dec 24 12:46:49 raspberrypi librespot[6210]: WARN:mdns::fsm: error sending packet Error { repr: Os { code: 99, message: "Cannot assign requested address" } }` is repeated when playing music. I haven't changed /etc/default/raspotify and I've scoured the online forums for solutions but haven't found anything. Has anyone had this issue or got any ideas for a solution?

Thanks in advance

dtcooper, 2017-12-27:
Can you send a little more info:
 * What version of raspbian?
 * What's your audio setup?
 * Is audio working with other applications, see anything muted in `alsamixer`?
 * Do you have pulseaudio installed?

morals100, 2017-12-28:
Really sorry! I was using my raspberry pi at a friend's house and it didn't work. I assume it was the DMZ, port forwarding or just their network setup that was messing it up. I went home and the error went away. Thanks though for responding!

dtcooper, 2017-12-28:
No problem! Glad you fixed it!

closed 2017-12-28 by dtcooper