dtcooper/raspotify #62: Raspotify crashed while listening (ERROR:librespot_core)

frederick0102, 2017-12-30:
Hi!

When I listened to music after 20-25 songs my Raspotify stopped with this error message in log:

```
INFO:librespot::player: Loading track "Rock or Bust"
ERROR:librespot_core::channel: channel error: 2 1
thread '<unnamed>' panicked at 'called `Result::unwrap()` on an `Err` value: ChannelError', /checkout/src/libcore/result.rs:860:4
note: Run with `RUST_BACKTRACE=1` for a backtrace.
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: "SendError(..)"', /checkout/src/libcore/result.rs:860:4
```

I didn't do anything just listened and it stopped. Restart in the command line helped to listen songs again, but I think this is a bug.

Anyway this is a fantastic work thank you for doing it!

Have a nice day!

dalori, 2018-01-24:
I do have a similar issue that after changing songs manually via phone or laptop. It works the first two to three times and than the audio output stops. The playback according to the Spotify clients continues. Only a reboot of the Raspberry Pi helps in that case. Is there a special location to where logs are written to so that I can debug this?

I only found the following in /var/log/user.log:

Jan 24 17:54:08 raspberrypi pulseaudio[976]: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Jan 24 17:54:08 raspberrypi pulseaudio[976]: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Jan 24 17:54:08 raspberrypi pulseaudio[976]: [pulseaudio] bluez4-util.c: org.bluez.Manager.GetProperties() failed: org.freedesktop.DBus.Error.UnknownMethod: Method "GetProperties" with signature "" on interface "org.bluez.Manager" doesn't exist

Nikodemski2002, 2018-02-17:
I have the same problem as dalori, but after 4 songs. Only command "sudo systemctl restart raspotify" and restarting spotify on my device helps. How i can fix it? Did someone solve this problem?

edoardesd, 2018-03-01:
Same problem for me, no solution :(

Nikodemski2002, 2018-03-01:
I have found a solution. I remove an program named audiopulse by command `sudo apt-get remove pulseaudio`. After that, everything work fine. :)

Sand0rf, 2018-05-09:
I too have this problem, however PulseAudio is not installed on my pi (Model 3, latest software and packages)

pi@raspberrypi:~ $ service raspotify status
● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2018-05-09 17:21:59 UTC; 30min ago
  Process: 1033 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 1030 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 1036 (librespot)
   CGroup: /system.slice/raspotify.service
           └─1036 /usr/bin/librespot --name SpotiPi --backend alsa --bitrate 320 --disable-audio-cache --enable-volume-normalisation --linear-volume --initial-volume=100

May 09 17:28:47 raspberrypi librespot[1036]: INFO:librespot_playback::player: Loading track "Stuck In The Middle With You" with Spotify URI "spotify:track:6pNVhq4m3S9iNRxpwmVQZm"
May 09 17:28:48 raspberrypi librespot[1036]: INFO:librespot_playback::player: Track "Stuck In The Middle With You" loaded
May 09 17:32:11 raspberrypi librespot[1036]: INFO:librespot_playback::player: Loading track "These Hands" with Spotify URI "spotify:track:67UV9f8OWjpwlmf9NuekTB"
May 09 17:32:11 raspberrypi librespot[1036]: INFO:librespot_playback::player: Track "These Hands" loaded
May 09 17:35:33 raspberrypi librespot[1036]: INFO:librespot_playback::player: Loading track "Perfect" with Spotify URI "spotify:track:0tgVpDi06FyKpA1z0VMD4v"
May 09 17:35:33 raspberrypi librespot[1036]: INFO:librespot_playback::player: Track "Perfect" loaded
May 09 17:39:56 raspberrypi librespot[1036]: INFO:librespot_playback::player: Loading track "Where Do You Go" with Spotify URI "spotify:track:2um4ISitgBDdMSZaiPyaIb"
May 09 17:39:57 raspberrypi librespot[1036]: INFO:librespot_playback::player: Track "Where Do You Go" loaded
May 09 17:40:11 raspberrypi librespot[1036]: ERROR:librespot_core::channel: channel error: 2 1
May 09 17:40:11 raspberrypi librespot[1036]: WARN:librespot_audio::fetch: error from channel

It happens at a random moment while playing. It can't see any specific event that triggers it. Any idea?

Bibo-Joshi, 2018-05-10:
Hey there,

I'm currently having the same problem.
It somehow started happening, after `raspotify` had been working fine for quite a while and I don't remember doing any upgrades oder updates … Also, `pulseaudio` ist **not** installed.
As far as I can guess from the `syslog`, it's more of a `librespot` problem than `raspotify`. In fact, it has already been mentioned in the [`librespot` issue #183](https://github.com/librespot-org/librespot/issues/183). Still, if someone has an idea, I'm glad for any advise :)
 
Here is a snap from the `syslog`:

```
May 10 13:55:23 openHABianPi librespot[1514]: ERROR:librespot_core::channel: channel error: 2 1
May 10 13:55:23 openHABianPi librespot[1514]: thread '<unnamed>' panicked at 'called 'Result::unwrap()' on an 'Err' value: ChannelError', libcore/result.rs:945:5
May 10 13:55:23 openHABianPi librespot[1514]: stack backtrace:
May 10 13:55:23 openHABianPi librespot[1514]:    0:   0x93a333 - std::sys::unix::backtrace::tracing::imp::unwind_backtrace::h4ef6490c84cef3d0
May 10 13:55:24 openHABianPi librespot[1514]:                        at libstd/sys/unix/backtrace/tracing/gcc_s.rs:49
May 10 13:55:24 openHABianPi librespot[1514]:    1:   0x93408f - std::sys_common::backtrace::_print::hea150ee4f33bb5de
May 10 13:55:24 openHABianPi librespot[1514]:                        at libstd/sys_common/backtrace.rs:71
May 10 13:55:24 openHABianPi librespot[1514]:    2:   0x938da3 - std::panicking::default_hook::{{closure}}::h6c4396ff5bd61d89
May 10 13:55:24 openHABianPi librespot[1514]:                        at libstd/sys_common/backtrace.rs:59
May 10 13:55:24 openHABianPi librespot[1514]:                        at libstd/panicking.rs:380
May 10 13:55:24 openHABianPi librespot[1514]:    3:   0x938987 - std::panicking::default_hook::h0f6cb16cf06a0198
May 10 13:55:24 openHABianPi librespot[1514]:                        at libstd/panicking.rs:396
May 10 13:55:24 openHABianPi librespot[1514]:    4:   0x939247 - std::panicking::rust_panic_with_hook::h2207195920ffa55f
May 10 13:55:24 openHABianPi librespot[1514]:                        at libstd/panicking.rs:576
May 10 13:55:24 openHABianPi librespot[1514]:    5:   0x9390ab - std::panicking::begin_panic::h8d43a09cf6d6e5f1
May 10 13:55:24 openHABianPi librespot[1514]:                        at libstd/panicking.rs:537
May 10 13:55:24 openHABianPi librespot[1514]:    6:   0x93902f - std::panicking::begin_panic_fmt::hd9afc94159ee9735
May 10 13:55:24 openHABianPi librespot[1514]:                        at libstd/panicking.rs:521
May 10 13:55:24 openHABianPi librespot[1514]:    7:   0x938fc7 - rust_begin_unwind
May 10 13:55:24 openHABianPi librespot[1514]:                        at libstd/panicking.rs:497
May 10 13:55:24 openHABianPi librespot[1514]:    8:   0x986e57 - core::panicking::panic_fmt::h0bd854df201d1baf
May 10 13:55:24 openHABianPi librespot[1514]:                        at libcore/panicking.rs:71
```

vincentkedison, 2018-10-14:
I'm currently having the same problem too. 
From the `syslog`:
```
Oct 14 01:36:02 luxodevpi librespot[7648]: INFO:librespot_playback::player: Loading track "Last Friday Night (T.G.I.F.)" with Spotify URI "spotify:track:3oHNJECGN3bBoGXejlw2b1"
Oct 14 01:36:07 luxodevpi librespot[7648]: ERROR:librespot_core::channel: channel error: 2 1
Oct 14 01:36:07 luxodevpi librespot[7648]: thread '<unnamed>' panicked at 'called `Result::unwrap()` on an `Err` value: ChannelError', libcore/result.rs:945:5
Oct 14 01:36:07 luxodevpi librespot[7648]: stack backtrace:
Oct 14 01:36:07 luxodevpi librespot[7648]:    0:   0x8ff60b - std::sys::unix::backtrace::tracing::imp::unwind_backtrace::hd0c8f132b6f079da
Oct 14 01:36:07 luxodevpi librespot[7648]:                        at libstd/sys/unix/backtrace/tracing/gcc_s.rs:49
Oct 14 01:36:07 luxodevpi librespot[7648]:    1:   0x908263 - std::sys_common::backtrace::print::h46c1c9484a27f45b
Oct 14 01:36:07 luxodevpi librespot[7648]:                        at libstd/sys_common/backtrace.rs:71
Oct 14 01:36:07 luxodevpi librespot[7648]:                        at libstd/sys_common/backtrace.rs:59
Oct 14 01:36:07 luxodevpi librespot[7648]:    2:   0x90195b - std::panicking::default_hook::{{closure}}::h2ee4c03f0d7d9285
Oct 14 01:36:07 luxodevpi librespot[7648]:                        at libstd/panicking.rs:211
Oct 14 01:36:07 luxodevpi librespot[7648]:    3:   0x9015bb - std::panicking::default_hook::hfc805264bfe1d452
Oct 14 01:36:07 luxodevpi librespot[7648]:                        at libstd/panicking.rs:227
Oct 14 01:36:07 luxodevpi librespot[7648]:    4:   0x90221f - std::panicking::rust_panic_with_hook::h9d57797cd21b21d7
Oct 14 01:36:07 luxodevpi librespot[7648]:                        at libstd/panicking.rs:511
Oct 14 01:36:07 luxodevpi librespot[7648]:    5:   0x901f57 - std::panicking::continue_panic_fmt::h376960e13d073506
Oct 14 01:36:07 luxodevpi librespot[7648]:                        at libstd/panicking.rs:426
Oct 14 01:36:07 luxodevpi librespot[7648]:    6:   0x901c03 - rust_begin_unwind
Oct 14 01:36:07 luxodevpi librespot[7648]:                        at libstd/panicking.rs:337
Oct 14 01:36:07 luxodevpi librespot[7648]:    7:   0x954da3 - core::panicking::panic_fmt::h0e6d5c6fb2a34dd6
Oct 14 01:36:07 luxodevpi librespot[7648]:                        at libcore/panicking.rs:92
Oct 14 01:36:07 luxodevpi librespot[7648]:    8:   0x50fdcb - core::result::unwrap_failed::hd61b5c0ce2095a05
Oct 14 01:36:07 luxodevpi librespot[7648]:    9:   0x4ecd33 - librespot_playback::player::PlayerInternal::run::he487512a17cfbd18
Oct 14 01:36:07 luxodevpi librespot[7648]:   10:   0x47d853 - std::sys_common::backtrace::__rust_begin_short_backtrace::hb7fc2c2002bd0e65
Oct 14 01:36:07 luxodevpi librespot[7648]:   11:   0x46d8cf - std::panicking::try::do_call::h3b4b9d2b7fdc0a9e
Oct 14 01:36:07 luxodevpi librespot[7648]:   12:   0x90f66b - __rust_maybe_catch_panic
Oct 14 01:36:07 luxodevpi librespot[7648]:                        at libpanic_unwind/lib.rs:105
Oct 14 01:36:07 luxodevpi librespot[7648]:   13:   0x4822eb - <F as alloc::boxed::FnBox<A>>::call_box::hfe946e389d5f9b5d
Oct 14 01:36:07 luxodevpi librespot[7648]:   14:   0x901003 - std::sys_common::thread::start_thread::ha977928eb5bc99e8
Oct 14 01:36:07 luxodevpi librespot[7648]:                        at /checkout/src/liballoc/boxed.rs:650
Oct 14 01:36:07 luxodevpi librespot[7648]:                        at libstd/sys_common/thread.rs:24
Oct 14 01:36:07 luxodevpi librespot[7648]:   15:   0x8ecd33 - std::sys::unix::thread::Thread::new::thread_start::h70dac9a9680028d2
Oct 14 01:36:07 luxodevpi librespot[7648]:                        at libstd/sys/unix/thread.rs:90
Oct 14 01:37:47 luxodevpi librespot[7648]: thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: "SendError(..)"', libcore/result.rs:945:5
Oct 14 01:37:47 luxodevpi librespot[7648]: stack backtrace:
Oct 14 01:37:47 luxodevpi librespot[7648]:    0:   0x8ff60b - std::sys::unix::backtrace::tracing::imp::unwind_backtrace::hd0c8f132b6f079da
Oct 14 01:37:47 luxodevpi librespot[7648]:                        at libstd/sys/unix/backtrace/tracing/gcc_s.rs:49
Oct 14 01:37:47 luxodevpi librespot[7648]:    1:   0x908263 - std::sys_common::backtrace::print::h46c1c9484a27f45b
Oct 14 01:37:47 luxodevpi librespot[7648]:                        at libstd/sys_common/backtrace.rs:71
Oct 14 01:37:47 luxodevpi librespot[7648]:                        at libstd/sys_common/backtrace.rs:59
Oct 14 01:37:47 luxodevpi librespot[7648]:    2:   0x90195b - std::panicking::default_hook::{{closure}}::h2ee4c03f0d7d9285
Oct 14 01:37:47 luxodevpi librespot[7648]:                        at libstd/panicking.rs:211
Oct 14 01:37:47 luxodevpi librespot[7648]:    3:   0x9015bb - std::panicking::default_hook::hfc805264bfe1d452
Oct 14 01:37:47 luxodevpi librespot[7648]:                        at libstd/panicking.rs:227
Oct 14 01:37:47 luxodevpi librespot[7648]:    4:   0x90221f - std::panicking::rust_panic_with_hook::h9d57797cd21b21d7
Oct 14 01:37:47 luxodevpi librespot[7648]:                        at libstd/panicking.rs:511
Oct 14 01:37:47 luxodevpi librespot[7648]:    5:   0x901f57 - std::panicking::continue_panic_fmt::h376960e13d073506
Oct 14 01:37:47 luxodevpi librespot[7648]:                        at libstd/panicking.rs:426
Oct 14 01:37:47 luxodevpi librespot[7648]:    6:   0x901c03 - rust_begin_unwind
Oct 14 01:37:47 luxodevpi librespot[7648]:                        at libstd/panicking.rs:337
[... cut ...]
Oct 14 01:37:47 luxodevpi librespot[7648]:    8:   0x50fb57 - core::result::unwrap_failed::h842e9a429b9016a8
Oct 14 01:37:47 luxodevpi librespot[7648]:    9:   0x4e48cf - librespot_playback::player::Player::load::h4bd8de08aadd8073
Oct 14 01:37:47 luxodevpi librespot[7648]:   10:   0x4abc0b - librespot_connect::spirc::SpircTask::load_track::h92d3edebd848d05b
Oct 14 01:37:47 luxodevpi librespot[7648]:   11:   0x4ab0cf - librespot_connect::spirc::SpircTask::handle_next::hf3b5fff88bcee548
```

I'm glad to hear someone have solution for this problem :)

closed 2021-11-24 by github-actions[bot]