dtcooper/raspotify #63: Cannot Use Bluetooth Speaker for Raspotify

zagingi, 2018-01-01:
I have my bluetooth speaker working on my Pi using raspotify(Tested with VLC and Music on my USB Drive) but I cannot find my bluetooth speaker on aplay -l or aplay -L. I have tried all the options with --device hw x,x in /etc/default/raspotify but no luck. 

Here is my output for aplay -l : 
`**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
`

maxfreedom97, 2018-01-10:
+1, can you please look at that, would be great feature!

danielsteiner, 2018-01-25:
Does the Bluetooth Speaker work with other Audio Software?

mountwebs, 2018-01-25:
Yes it does. The difference is that the audio from raspotify/librespot does
not pass through alsamixer or pulseaudio, and is therefore not affected by
the choice of device. I have not figured out how - or if - librespots
--device command can be set to the bluetooth device.

mountwebs, 2018-01-25:
I got it to work when I compiled librespot myself with alsa as backend.

romano-w, 2018-03-05:
I have been working on the same problem for hours now, and I have finally found this issue but do not know how to compile alsa with raspotify myself. How did you do that?

valmitico, 2018-03-06:
I've a bluetooth speaker connect to my rpi3 and I don't be able to listen raspotify via bluetooth. please......

mountwebs, 2018-03-07:
I used librespot directly instead of raspotify - so it doesn't really belong here. You can find instruction in the readme https://github.com/plietar/librespot and https://github.com/librespot-org/librespot/wiki/Compiling. I think i compiled with "cargo build --release --no-default-features --features alsa-backend".

HappyLuke, 2018-04-23:
Hi all,
I got raspotify running with a bluetooth device. Need to google a lot and found that no one had a real solution. So here it is:

1. Connect your bluetooth speaker with `bluetoothctl`, if not done yet.
1.1. `bluetoothctl`
1.2. `agent on`
1.3. `pairable on`
1.4. `scan on` and find your bluetooth device, copy the MAC
1.5. `scan off`
1.6. `pair XX:XX:XX:XX:XX:XX` (insert the MAC)
1.7. `trust XX:XX:XX:XX:XX:XX`
1.8. `connect XX:XX:XX:XX:XX:XX`
1.9. `exit`
2. restart your pi `sudo shutdown -r now`
3. you have to configure your audio output in /etc/asound.conf (~/.asoundrc is not working) with dynamic and static ALSA configuration:
`defaults.bluealsa {
     interface "hci0"            # host Bluetooth adapter
     device "XX:XX:XX:XX:XX:XX"     # Bluetooth Device MAC
     profile "a2dp"
}
pcm.yourdevice {
        type plug
        slave.pcm {
                type bluealsa
                device "XX:XX:XX:XX:XX:XX"
                profile "a2dp"
        }
        hint {
                show on
                description "Your description of the bluetooth device"
        }
}`
4. Configure raspotify for using your bluetooth device (here you can use either 'bluealsa' or 'yourdevice' as device). Find the line starting with `#OPTIONS` and replace it with following line:
`OPTIONS="--device bluealsa"`
5. Restart raspotify `sudo service raspotify restart`

Now your raspotify should be able to play your music on the specified bluetooth device!

Hope this helps and have fun with your pi :)

mountwebs, 2018-04-24:
Great work, HappyLuke!

Is this possible now because of this new fix? https://github.com/dtcooper/raspotify/issues/83

HappyLuke, 2018-04-26:
Hi mountwebs,

don't think so.. The fix just enables the possibility to configure the backend. However, I use the default backend, which is alsa, so the fix shouldn't have changed anything. It should have been possible a I described even before.

mountwebs, 2018-04-27:
I see! I will try your solution!

janekszynal, 2019-01-16:
Thanks @HappyLuke for the initial solution and @tinoargentino for pasting the configuration, it worked immediately for me!

jimmylpx, 2019-02-23:
> @Zumzod I made it work writing this in the asound.conf file:
> 
> ```
> defaults.bluealsa.interface "hci0"
> defaults.bluealsa.device "XX:XX:XX:XX:XX"
> defaults.bluealsa.profile "a2dp"
> 
> pcm.btheadset {
>     type plug
>     slave {
>         pcm {
>               type bluealsa
>               device XX:XX:XX:XX:XX:XX 
>               profile "auto"
>          }   
>     }   
>     hint {
>          show on
>          description "BT Headset"
>     }   
> }
> ctl.btheadset {
>     type bluetooth
> } 
> ```
> good luck!

thx mate,,,you are my hero haha

efrenefren, 2019-04-30:
if you are constantly switching between audio devices it would be best to symlink asoundrc to asound.conf
`rm /etc/asound.conf
ln -s /home/pi/.asoundrc /etc/asound.conf`

then restart raspotify everytime you change device
`sudo service raspotify restart`

RaspDaniel, 2019-05-21:
Thank you guys it very helped me!!!! :DD

djorr5, 2019-06-20:
Hi all, sorry to bring up an old thread. I am running Stretch and don't have the asound.conf file in /etc. Is it just a case of creating this file with the code @tinoargentino made?

PolFrank, 2019-10-02:
Awesome! Thanks!

After modifying /etc/asound.conf as mentioned earlier, I can switch between Bt-headset and analog by editing the OPTIONS line in the file /etc/default/raspotify

OPTIONS="--device bluealsa" 
#to play from the BT-headset

or

#OPTIONS="--device bluealsa"
#to play from the analog audio output

Don't forget to always:
sudo service raspotify restart
after every change.

Kerbo1, 2020-02-24:
@HappyLuke @tinoargentino you guys are legends, I would've never even come close to this

xtools-at, 2020-03-28:
for everybody still struggling, i've found this to be installed alongside raspotify resolving any issues for me: https://github.com/bablokb/pi-btaudio
(remove pulseaudio if you have it)

mabittar, 2020-04-08:
> for everybody still struggling, i've found this to resolve any issues for me when installed alongside raspotify : https://github.com/bablokb/pi-btaudio
> (remove pulseaudio if you have it). still using the default alsa backend.

thank you..... it´s easy and fast to do it

Wintersss1, 2020-06-22:
Hi, i cant get it working... maybe somebody has time to help me? 
HW: Raspberry Pi 4B
Bluetooth Device: Teufel Rockster Air

skywinder, 2020-08-29:
> @Zumzod I made it work writing this in the asound.conf file:
>
> ```
> defaults.bluealsa.interface "hci0"
> defaults.bluealsa.device "XX:XX:XX:XX:XX"
> defaults.bluealsa.profile "a2dp"
>
> pcm.btheadset { [...]

Thank you! It work! @dtcooper would we pin in to readme? looks it's quite popular thing. It could save hours for next users.

dtcooper, 2020-08-29:
Sure. Submit a PR and I'll merge into the readme.

closed 2020-10-18 by dtcooper

gbcatrinoiu, 2021-01-31:
> for everybody still struggling, i've found this to resolve any issues for me when installed alongside raspotify : https://github.com/bablokb/pi-btaudio
> (remove pulseaudio if you have it). still using the default alsa backend.

This works like a charm!

michelefa1988, 2022-02-27:
I ran into the issue above running a CLEAN version of Raspberian PI BUSTER. What is the best approach to run in February 2021? I previously had RASPOTIFY working with BlueAlsa, but I am wondering if anything has changed since previous OSs.

My goal is to enable sound over bluetooth and I do not want to try any methods which could be outdated on the latest Buster.

Thanks

JasonLG1979, 2022-02-27:
@michelefa1988 please do not necrobump closed issues especially in this case since Bluetooth is completely out of scope of Raspotify. Bluetooth, bluealsa and whatnot are much broader issues that we have no control over. Solutions for that particular use-case would be generally applicable to any situation where you'd want to broadcast or receive Bluetooth audio with a Pi. You would have better luck elsewhere.

PR #303 (2020-10-18): add missing manual for bluetooth connection
will close #63
commit c07593a7 (2020-10-18): add missing manual for bluetooth connection

will close #63