dtcooper/raspotify #65: Spotify Connect found, but cannot play music

MaChristopher, 2018-01-07:
Hello,

since today I have the following issue, without having changed anything on my setup. Spotify Connect on the Raspberry Pi with installed raspotify is found in Spotify (Android App and iOS Application). However, it is greyed out and cannot be selected. I tried restarting the service, also renaming the device. The device is always found under the current selected name(configured in /etc/default/raspotify), however it can never be selected. I completely uninstalled raspotify according to README and installed it back again. The behavior remains the same. 

[image]

I did already several reboots of the raspberry pi or manual restart of service

pizzaboy75, 2018-01-09:
This is normal behaviour if you try to play podcasts. Try a normal music album, it should work then.

dtcooper, 2018-01-09:
Unfortunately, this is due to librespot. Closing for now. If you find a fork that supports podcasts, please let me know on this ticket and I'll consider building from that.

closed 2018-01-09 by dtcooper