dtcooper/raspotify #67: Volume low at 100%

MrJoki007, 2018-01-13:
Before i used Spotify Connect Web but that does not work anymore in Debian stretch.
Now i upgraded to Raspotify but the Problem i have is that max Volume on Spotify Connect Web was very loud but now when i turn it to max its like a quarter of the old volume.

dalori, 2018-01-24:
This is not an issue concerning the repo, is it? I suggest to add a remark into the readme to use e.g. `alsamixer` in the command line to increase the volume of the Raspberry Pi.

MrJoki007, 2018-01-24:
Oh yes that was the problem.
Ive seen alot of users who had this problem

closed 2018-01-24 by MrJoki007

dtcooper, 2018-01-25:
Duly noted. Thanks for helping @MrJoki007, @dalori. I'll create a Troubleshooting section to the Readme and add this there. :)