dtcooper/raspotify #69: Enable Scrobbling

teamreflect, 2018-01-19:
Is there a way to enable scrobbling for raspotify? I'm trying to intergrate it to another raspberry pi project called MagicMirror so that it will display the track information on the screen while raspotify runs and play songs in the background.

krzysztofkarolak, 2018-01-29:
I've also had the same problem with scrobbling. You can get the information about current track by calling a request to Spotify API associated with your Spotify account. If you need to scrobble to last.fm, you can connect your Spotify account with last.fm profile and current track info will be grabbed automatically.

notABluesSinger, 2018-05-11:
This may be a non-issue now. Last.fm very recently changed to an integrated setup with Spotify. You may have got an email... if not, visit https://www.last.fm/settings/applications and see if you can configure it there. Hope this helps :)

closed 2021-11-24 by github-actions[bot]