dtcooper/raspotify #74: Spotify Behavior Question

soilstack, 2018-02-06:
Would someone mind testing:

1) play music through a raspotify-device selected from spotify on an iphone
2) music playing out of that device now
3) record video from the iphone while music is playing
4) does iphone pause music from the raspotify device while recording video?

Thank you.

VanHinten, 2018-02-07:
Hey. I hope i've understood your question. 

If you select a song with your phone, the raspberry gets the song directly from spotify - not from your phone. (You can test this by closing spotify on your phone - with spotify connect the song still plays).

So the phone is not streaming music (and the camera can't stop the music from playing tho).

Hope this answers your question.
You can read more about that at the spotify connect webpage (tag: 'Safe Battery Life')

soilstack, 2018-02-08:
Oh ok, great.  I didn't realize this - -thought it just streamed like a speaker.  Thanks for the clarification.

closed 2018-02-08 by soilstack