dtcooper/raspotify #78: How to get it working with pulseaudio

dustXman, 2018-02-10:
Hi there,

I guess I have a noobee 🐝 question:
I want to run raspotify with pulseaudio (which is supposed to stream the output then as rtp-server throughout my network - but thats a story for another day) 

When I try running the `librespot` in `/usr/share` that comes with it with "`--backend pulseaudio`" it fails with "`backend invalid`" . I guess I have to rebuild librespot - right? Is this possible with my Pi 3 directly? Or how to do so?

tomkcook, 2018-07-11:
Bump.  Same issue here.

dtcooper, 2018-07-11:
Have you tried just using as-is? What results do you get?

atchoo78, 2019-01-19:
Just wanted to share my experience, for what it's worth. As usual when I debug or try to make something work, I can't always pinpoint exactly the things that made the difference, but on this one I have a pretty good idea. 

I was having major issues with PulseAudio errors and X11-related error loops, until I did this:

- `sudo systemctl stop raspotify.service`

- `pulseaudio -k`

- removed the "device" and "backend" options/arguments from "ExecStart" in `/lib/systemd/system/raspotify.service`

- Edited /etc/default/raspotify and added `OPTIONS="--username MYUSERNAME  --password MYPASSWORD"`, as well as the volume options. Commented out `BACKEND_ARGS="--backend alsa"`.

- `pulseaudio -D` and `pulseaudio --start` (probably not necessary to run both though)

    - (To get my Bluetooth receiver into the loop at this point: `bluetoothctl`, `connect <mac adress of BT receiver>` and `exit`)

- `sudo systemctl start raspotify.service`
- Started Spotify, chose the Spotify Connect Output device, and played a song. Got Bluetooth audio output in 320 kbs. Checked the output of `sudo systemctl status raspotify.service`. No errors whatsoever.

## Conclusion/Status
Whenever I include arguments regarding device output, source, server etc., I get all kinds of permission errors regarding Pulseaudio, most likely because of a missing $DISPLAY for X11 because I mostly use SSH logins. 

For the moment, Raspotify is very stable with very little overhead/latency, _as long as I don't bring any special sound device configurations to the party,_ and let Raspotify figure that out on its own :-)

Thanks for making this!

Cheers,

Andreas Lorentsen
🌞

 

 

-

TeaZee, 2020-02-08:
Man, I would love to have raspotify working with PulseAudio to redirect the audio to airplay/airtunes (i.e. my Pioneer receiver which is no longer supported by Spotify...)

I have been trying to get this to work using many many suggestions on the web, but it just won't. I would sound logical that the "pulseaudio" support should be set at compile time, but some seem to get it to work without (BUT HOW!!).

Pulseaudio will give me my airtunes reciever as sink. But I do not have a source. So raspotify plays on the Raspberry jack. <Sigh...>

Anyone?

profwyattb, 2020-02-09:
I would also love a version with pulseaudio support. Alsa is so difficult to configure!

sniperrifle2004, 2020-02-28:
I got to a solution without modifying librespot (leaving the direct pulse support to hopefully some future version of the rodio backend (or more specifically cpal)).

It turned out that, regularly, programs with alsa playback (in this case librespot) get access to pulse through the alsa pulse plugin (This is actually currently the default alsa device's behaviour for my pi at least). However since the raspotify service is run in isolation at the system level using the raspotify user it can't access any pulseaudio service (which is a user level service) so it doesn't work. I have created a user level modified version of the raspotify service and use that instead and then it just works as it should.

profwyattb, 2020-02-28:
Thanks @sniperrifle2004, Did you just modify the service to have it run as pi user or did you just do a crontab script?

sniperrifle2004, 2020-02-28:
Neither I copied it into `~/.config/systemd/user/`, and modified for that context. This is what I ended up with taking out what would be problematic or redundant:

```
[Unit]                                                                                                                                                                                                                                                                                                                     
Description=Raspotify                                                                                                                                                                                                                                                                                                      
After=network.target                                                                                                                                                                                                                                                                                                       
                                                                                                                                                                                                                                                                                                                            
[Service]                                                                                                                                                                                                                                                                                                                  
Restart=always                                                                                                                                                                                                                                                                                                             
RestartSec=10                                                                                                                                                                                                                                                                                                              
Environment="DEVICE_NAME=raspotify (%H)"                                                                                                                                                                                                                                                                                   
Environment="BITRATE=160"                                                                                                                                                                                                                                                                                                  
Environment="CACHE_ARGS=--disable-audio-cache"                                                                                                                                                                                                                                                                             
Environment="VOLUME_ARGS=--enable-volume-normalisation --linear-volume --initial-volume=100"                                                                                                                                                                                                                               
Environment="BACKEND_ARGS=--backend alsa"                                                                                                                                                                                                                                                                                  
EnvironmentFile=-/etc/default/raspotify                                                                                                                                                                                                                                                                                    
ExecStart=/usr/bin/librespot --name ${DEVICE_NAME} $BACKEND_ARGS --bitrate ${BITRATE} $CACHE_ARGS $VOLUME_ARGS $OPTIONS                                                                                                                                                                                                    
                                                                                                                                                                                                                                                                                                                            
[Install]                                                                                                                                                                                                                                                                                                                  
WantedBy=default.target
```

profwyattb, 2020-02-28:
So this should work if I have another audio service running on the Pi that uses pulseaudio, will they play nice? 🤞

sniperrifle2004, 2020-02-29:
Yes, they should and I have been able to confirm that to some extent through some testing. I can't guarantee there won't be any problems, but I haven't run into any so far.

dtcooper, 2020-02-29:
If one of you folks want to submit a PR updating the README, I'd be happy to accept. :)

b0bbywan, 2020-05-02:
Hey I found this issue to easily use raspotify with pulseaudio https://github.com/dtcooper/raspotify/issues/154

andrewfedun, 2020-06-06:
> Neither I copied it into `~/.config/systemd/user/`, and modified for that context. This is what I ended up with taking out what would be problematic or redundant:
>
> ``` [...]

@sniperrifle2004 I'm having a little trouble understanding what you mean here. 

1) By ~/.config/systemd/user/ is that referring to /home/pi/.config/systemd/user/ ? If so, that doesn't currently exist in my filesystem so are you saying we should create that directory?
2) When you say "I copied it into..." what is it referring to? 

thanks for your help!

profwyattb, 2020-06-06:
Hey, I got this working following @sniperrifle2004 instructions.

1. Copy either the code @sniperrifle2004 pasted in here or the original systemd file (your mileage may vary) into /home/pi/.config/systemd/user/ (~ refers to the users home directory). If the directory doesn't exist, create it
2. Run `sudo systemctl **--user** enable <name of file copied/created>`
3. Profit?!?

Hope this helps :)

andrewfedun, 2020-06-06:
Hi @profwyattb , thanks for the help i really appreciate it. So i went as far as /home/pi/.config but it didn't have the folder systems within config. So I created systemd and user and made a file called "usethis" where I pasted sniperrifle2004's code. Then, when I tried to run sudo systemctl --user enable usethis , it said  Failed to connect to bus: No such file or directory. Here's a screenshot of the terminal, please let me know if you know how I can fix this. Thanks again for all your help, I really appreciate it.

[image]

profwyattb, 2020-06-06:
Sorry that was my bad, run `systemctl --user enable <name of file copied/created>` without `sudo`. Have edited original comment. Also, you don't need to type the whole path if you're already in the directory. You can just type the filename

andrewfedun, 2020-06-06:
@profwyattb awesome I think that worked. do you know how to make it find raspotify as a source? when I run pactl list sources, it's still only giving me the analog and the HDMI

profwyattb, 2020-06-06:
Run `systemctl --user start <name of file copied/created>` and try again. What is your use case for this btw?

andrewfedun, 2020-06-06:
so I made an audio visualizer using CAVA and running the FIFO output into a python program, which then feeds it to a 16x32 matrix using the hzeller library. It works great when I'm playing local mp3 or anything on youtube, but it doesn't work with raspotify because CAVA uses pulse, which isn't recognizing raspotify output right now. Here's a link to a quick demo [video](https://drive.google.com/drive/u/0/folders/1fS_PoDZgUl95EvfNS4aMVkYnsXxbwOXm) if you're interested:

i ran that command again and rebooted but I'm still only getting two sources. In yours when you run pactl list sources do you get a source specific to raspotify?

profwyattb, 2020-06-06:
@andrewfedun That looks sick! I don't know what the command outputs as I don't have anything running it right now. Try playing a song through it and see what happens when you list sources. Otherwise, I'm sure there are some people here who would run that command for you :)

andrewfedun, 2020-06-06:
@profwyattb thanks man! that hasn't worked for me but I'll keep trying things and update this thread if I figure it out.

@sniperrifle2004 if you or anyone else has been able to get raspotify to show up as a source from the command **pactl list sources** I would literally venmo you to show me how

profwyattb, 2020-06-06:
@andrewfedun If I have the time, I **might** be able to check it for you.

andrewfedun, 2020-06-06:
@profwyattb awesome, sounds good just let me know

andrewfedun, 2020-06-06:
@profwyattb wanted to update & clarify this a little bit since my other comments weren't fully accurate. I said I need raspotify to show up as a source but that's not how pulseaudio works. The source is just alsa_output.platform-soc_audio.analog-mono.monitor and it already shows up - that's just monitoring whatever goes through the 3.5 jack. The issue is that even though raspotify is playing out of the 3.5 jack, its not seen by pulseaudio so my audio visualizer basically flatlines. I need pulse to recognize the raspotify coming out of the 3.5 jack so it shows up when using the source alsa_output.platform-soc_audio.analog-mono.monitor.

andrewfedun, 2020-06-10:
I finally got it working - wanted to put an update in this thread.

After grinding through all the various solutions of getting raspotify to work with pulse, the best I could get working was a super corrupted demonic sounding version of the songs I was playing. There's a thread that explains how to fix the corrupted sound, but that wasn't working for me either. After all that I decided to give the original librespot a try and it worked like a charm. It's not that much harder to build out than raspotify and it's much easier to tailor to specific needs. For anyone that needs to get pulse working, I'd recommend just using librespot with a pulseaudio backend. I love raspotify though and it is a great solution for anyone who just needs Spotify to work.

profwyattb, 2020-06-10:
Hey @andrewfedun,

I'm glad you finally got it working.  Linux audio is a hot mess - I've spend many hours messing with config files. 

How much did the cost of the project work out to be - I'm really interested in the project and was thinking of building my own. Did you follow a guide or go it alone?

Thanks :)

TeaZee, 2020-06-11:
Thanks @andrewfedun,
I see you did nothing special beyond run the instructions from librespot and succeeded... I was not happy to install all kind of new compiler software (rust, cargo, ...) but your success made me do it. :-)

It's compiling right now on a raspberry pi2 (temp stays  around 52*C without active cooling (if internal measure us accurate) with 4 cores on 100% load. I executed with the `time` command, so I can benchmark your "forever" ;-)

Thx.

TeaZee, 2020-06-11:
Ok. It took 43 minutes, temp stayed below 54*C (at 80*C the should interfere and start throttling, so that is cool).
But... after 43 minutes the build failed on linking `pulse` and `pulse-simple`. So maybe you left out something on your recipe?

andrewfedun, 2020-06-11:
@TeaZee hmm sorry to hear it didn't cross the finish line. Unfortunately, those were the exact steps I used, I wrote down each one in a google doc as I did them so that really is everything I did. I would check out the issues page of librespot and see if anyone else had your error.

TeaZee, 2020-06-11:
Because it is so obvious, it was hard to find:
You mention installing `libasound2-dev` twice in your recipe... (at 1. and 3.)
But you missed `libpulse-dev` (And pulse is what we are talking about here!) `apt install libpulse-dev` did the trick. (maybe `libasound2-dev` is not needed at all (Alsa?)

Anyway 1:30 extre compile & link time and finished. Now see if I can get it to connect to Pulse and stream AirPlay to the amplifier...

andrewfedun, 2020-06-11:
@TeaZee ahh yeah I mustve had that already installed and didn't realize it would be a necessary step. glad you got it - best of luck with the airplay stuff.

@dtcooper to be honest I don't understand the process of a PR at all. I could definitely write out step-by-step instructions as a word document for those looking to get pulseaudio working. If I post that file in here could you add that to the readme?

TeaZee, 2020-06-11:
Ok. Librespot works fine. PulseAudio (version precompiled package) does not support AirPlay properly (needs raop2 i suspect). But PulseAudio-DNLA does a similar job of playing the audio on my (DNLA) devices. And that seems to work...

sniperrifle2004, 2020-06-11:
@andrewfedun Nice that using the pulseaudio backend works for you!

Anyway, it turns out I also had sound corruption problems. I initially missed this problem entirely due to it only happening on the internal audio and not my bluetooth speakers which I was using for raspotify (Which was also my motivation for using pulse over alsa).

I wonder are you running on buster? Because if so, I managed to fix the sound corruption when using the alsa backend - After convincing myself it should work ie. the combination of alsa and pulse wasn't the problem - by instead of building against alsa-lib 1.0.29, building against alsa-lib-1.1.8 which is the version of alsa-lib actually on my pi. It may be a very subtle incompatibility somewhere in the ABI. Unfortunately I don't really have an idea how to fix this in general though. My 'fixed' Dockerfile would likely break stretch builds.

andrewfedun, 2020-06-11:
@sniperrifle2004 Yes I am on buster - that's a very cool fix and I'm glad you were able to get it to work! I'm settled into my librespot setup so I can't test that out on mine but if anyone else comes across this and can verify I'm sure it would be very helpful to some poor soul struggling with the mess of Linux audio

digixmax, 2021-03-26:
> 
> 
> Hey I found this issue to easily use raspotify with pulseaudio #154

Great find!

chandlercv, 2021-04-01:
Hi - I finally got things working on raspbian buster with desktop due to this insight from @sniperrifle2004 
```
. . . [because] the raspotify service is run in isolation at the system level using the raspotify user it can't access any pulseaudio service (which is a user level service) so it doesn't work.
```

I still hit a couple of snags getting it to work, but I was able to persevere and overcome, and I'd like to put this into a PR for the readme.

The only snag that wasn't due to me needing to learn more about systemd was the Raspotify service was not starting on its own after login.

I finally got it to work, and the last thing I changed before it started working was I changed the [Unit] section to read:
```
[Unit]
Description=Raspotify
Wants=pulseaudio.service
```

then ran `systemctl --user reenable raspotify.service`

I was definitely intending to make this wait until the pulseaudio.service was running.  Also, when I checked `systemctl --user -t target` I did not see a network.target listed there.

Was this change necessary?  Or is it more likely that something I didn't realize I did made this all work?

dtcooper, 2021-04-05:
Check if the commit I just merged explains how to fix this. Closing for now, let me know if this fixes your problems. Thanks!

closed 2021-04-05 by dtcooper