dtcooper/raspotify #82: Problem with 3.5mm jack output

chongkim97, 2018-02-19:
I had installed everything without error but whenever I would try and play something it would come out as static, would anyone have any idea as to why it happens or how I could solve it

brunob45, 2018-02-19:
Are you using an HDMI output? The RPi could be outputting the sound there.. Try forcing the audio output to the jack.
$ amixer cset numid=3 1
or
$ sudo raspi-config
then select Advanced Options -> Audio -> 1 Force 3.5 mm, then Press Enter
Use the right arrow to select Finish, then Enter.

chongkim97, 2018-02-19:
Yeah, I've forced the output towards to the jack

enapupe, 2018-02-26:
Try a different p2 connector, worked for me. From loud static noise to clean output.

michal2616, 2018-03-11:
My problem seems to be simillar. I used to use raspotify for few months. One day it just stopped working. I mean raspotify was on but music wasn't playing. Quick removal and install again. Music plays again except I can hear statics. Weird thing is that when I turn (on my mobile spotify client) volume down to 0% I can still hear music (very quietly) playing in raspotify. I checked if maybe 2 raspotify clients are up on my raspberry but that is not true. Still didn't find any cause of the problem

ondrejtichacek, 2018-09-04:
Any updates on this issue? I had same problems as @chongkim97, Pi(v1)B, static highly correlated to wifi activity. I only managed to solve it by using an external (usb) sound card.

chm-dev, 2021-01-19:
I was able to make raspotify play via jack by changing raspotify config file.
I have a Raspberry Pi 2 running Debian 10 with HDMI cable disconnected at all times. No window env.
I couldn't change default output from HDMI to Analog (Headphones). Even adding` hdmi_ignore_edid_audio=1` in config.txt made no effect.

Forced analog output by adding below line to `/etc/default/raspotify`
OPTIONS="--device hw:1,0"

dcambria, 2021-01-23:
@chm-dev thanks. It works now :)

The-real-Hezarfen, 2021-03-10:
@chm-dev  Works just fine, thanks!

piotr-cz, 2021-09-02:
More info here: https://github.com/dtcooper/raspotify/pull/386

closed 2021-11-24 by JasonLG1979