dtcooper/raspotify #83: --backend alsa prevents --backend pipe to work

snizzleorg, 2018-02-21:
in ```raspotify/raspotify/lib/systemd/system/raspotify.service```
libresport gets called with 
```
ExecStart=/usr/bin/librespot --name ${DEVICE_NAME} --backend alsa --bitrate ${BITRATE} $CACHE_ARGS $OPTIONS
```

This effectively prevents the user from using a different backend (in my case pipe). 

The backend should be configurable option.

I can pass the parameter in ```/etc/default/raspotify```

like so: 

```
OPTIONS="--username USER --password PQSS --backend pipe --device /home/pi/music/spotify"
```

but then the deamon will complain since the option ```--backend``` will be passed twice.

So i suggest the following:
```raspotify/raspotify/lib/systemd/system/raspotify.service```:
```
[Unit]
Description=Raspotify
After=network.target

[Service]
User=raspotify
Group=raspotify
Restart=always
RestartSec=10
PermissionsStartOnly=true
ExecStartPre=/bin/mkdir -p 0755 /var/cache/raspotify ; /bin/chown raspotify:raspotify /var/cache/raspotify
Environment="DEVICE_NAME=raspotify (%H)"
Environment="BITRATE=160"
Environment="CACHE_ARGS=--disable-audio-cache"
EnvironmentFile=-/etc/default/raspotify
ExecStart=/usr/bin/librespot --name ${DEVICE_NAME} --backend $BACKEND --bitrate ${BITRATE} $CACHE_ARGS $OPTIONS

[Install]
WantedBy=multi-user.target
```

and then 
```/etc/default/raspotify```:
```
# /etc/default/raspotify -- Arguments for librespot

# Device name on Spotify Connect
#DEVICE_NAME="raspotify"

# Bitrate, one of 96 (low quality), 160 (default quality), or 320 (high quality)
#BITRATE="160"

# Additional command line arguments for librespot can be set below.
# See `librespot -h` for more info.
#
# To make your device visible on Spotify Connect across the Internet add your
# username and password which can be set via "Set device password", on your
# account settings, use `--username` and `--password`.
#
# To choose a different output device (ie a USB audio dongle or HDMI audio out),
# use `--device` with something like `--device hw:0,1`. Your mileage may vary.
#
OPTIONS="--username USER --password PASS  --device /home/pi/music/spotify"
BACKEND="pipe"
#BACKEND="ALSA"

# Uncomment to use a cache for downloaded audio files. It's best to leave this
# as-is since permissions are properly set on directory `/var/cache/raspotify'.
#CACHE_ARGS="--cache /var/cache/raspotify"
```

LadnerJamie, 2018-03-08:
I ran into this issue as well however in raspotify.service I changed

--backend $BACKEND  to be --backend ${BACKEND}

dtcooper, 2018-03-08:
Thanks for filing. I'll add something like `BACKEND=--backend alsa` to the `/etc/default/raspotify` file so you can customize this without having to mess with the [systemd](https://www.freedesktop.org/wiki/Software/systemd/) config directly.

dtcooper, 2018-03-18:
Also considering creating a pulse enabled build. Hmmmm.

commit b3503c44 (2018-03-24): Allow for alternate backends, such as pipe. Fixes #83
closed 2018-03-24 by dtcooper