dtcooper/raspotify #84: raspotify ALWAYS stops working after second track

layouterlimits, 2018-02-22:
reliably after second track the raspotify-spot disappears from devices. The service still seems to be running though:

`root@Kueche:~# service raspotify status
● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2018-02-22 00:55:46 CET; 8h ago
  Process: 457 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 437 ExecStartPre=/bin/mkdir -p 0755 /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 462 (librespot)
   CGroup: /system.slice/raspotify.service
           └─462 /usr/bin/librespot --name raspotify (Kueche) --backend alsa --bitrate 160 --disable-audio-cache

Feb 22 09:00:26 Kueche librespot[462]:    8:   0x8a945f - <unknown>
Feb 22 09:00:26 Kueche librespot[462]:    9:   0x8ae13b - <unknown>
Feb 22 09:00:26 Kueche librespot[462]:   10:   0x4b2697 - <unknown>
Feb 22 09:00:26 Kueche librespot[462]:   11:   0x4c9403 - <unknown>
Feb 22 09:00:26 Kueche librespot[462]:   12:   0x4b29ab - <unknown>
Feb 22 09:00:26 Kueche librespot[462]:   13:   0x8c95af - <unknown>
Feb 22 09:00:26 Kueche librespot[462]:   14:   0x8f027f - <unknown>
Feb 22 09:00:26 Kueche librespot[462]:   15:   0x8d0a2f - <unknown>
Feb 22 09:00:26 Kueche librespot[462]:   16:   0x4c9e5f - main
Feb 22 09:00:26 Kueche librespot[462]:   17: 0xb6cc5677 - __libc_start_main`

geret13, 2018-02-24:
I am having the same issue, maybe this stacktrace might be of some better use:
```
Feb 24 00:33:45 raspberrypi librespot[1816]: thread 'main' panicked at 'Box<Any>', core/src/session.rs:87:43
Feb 24 00:33:45 raspberrypi librespot[1816]: stack backtrace:
Feb 24 00:33:45 raspberrypi librespot[1816]:    0:   0x87ce3b - <unknown>
Feb 24 00:33:45 raspberrypi librespot[1816]:    1:   0x886f93 - <unknown>
Feb 24 00:33:45 raspberrypi librespot[1816]:    2:   0x8754f3 - <unknown>
Feb 24 00:33:45 raspberrypi librespot[1816]:    3:   0x874fcf - <unknown>
Feb 24 00:33:45 raspberrypi librespot[1816]:    4:   0x87599f - <unknown>
Feb 24 00:33:45 raspberrypi librespot[1816]:    5:   0x584973 - <unknown>
Feb 24 00:33:45 raspberrypi librespot[1816]:    6:   0x5ffd8b - <unknown>
Feb 24 00:33:45 raspberrypi librespot[1816]:    7:   0x856a97 - <unknown>
Feb 24 00:33:45 raspberrypi librespot[1816]:    8:   0x8554f7 - <unknown>
Feb 24 00:33:45 raspberrypi librespot[1816]:    9:   0x85a1d3 - <unknown>
Feb 24 00:33:45 raspberrypi librespot[1816]:   10:   0x45e697 - <unknown>
Feb 24 00:33:45 raspberrypi librespot[1816]:   11:   0x475403 - <unknown>
Feb 24 00:33:45 raspberrypi librespot[1816]:   12:   0x45e9ab - <unknown>
Feb 24 00:33:45 raspberrypi librespot[1816]:   13:   0x875647 - <unknown>
Feb 24 00:33:45 raspberrypi librespot[1816]:   14:   0x89c317 - <unknown>
Feb 24 00:33:45 raspberrypi librespot[1816]:   15:   0x87cac7 - <unknown>
Feb 24 00:33:45 raspberrypi librespot[1816]:   16:   0x475e5f - main
Feb 24 00:33:45 raspberrypi librespot[1816]:   17: 0xb6d8e677 - __libc_start_main
```

catercousin, 2018-02-24:
Same issue, same stacktrace as @geret13 happens every 30min - 1hour.

layouterlimits, 2018-02-27:
ah ok...that seems to be exact the same issue..:

`root@Kueche:~# librespot --name 'TEST'
INFO:librespot: librespot (raspotify) 8e0d758 (2018-02-26). Built on 2018-02-26. Build ID: U2Bp36G1
INFO:librespot_core::session: Connecting to AP "gew1-accesspoint-b-h3rm.ap.spotify.com:4070"
INFO:librespot_core::session: Authenticated as "n84a4vpcb5zepta6sm3y3i99h" !
INFO:librespot_core::session: Country: "DE"
INFO:librespot_playback::audio_backend::alsa: Using alsa sink
INFO:librespot_playback::player: Loading track "Demon Cleaner"
INFO:librespot_playback::player: Track "Demon Cleaner" loaded
INFO:librespot_playback::player: Loading track "Losing All"
INFO:librespot_playback::player: Track "Losing All" loaded
thread 'main' panicked at 'Box<Any>', core/src/session.rs:79:43
stack backtrace:
   0:   0x8c7057 - <unknown>
   1:   0x8d12bb - <unknown>
   2:   0x8bf8d7 - <unknown>
   3:   0x8bf3b3 - <unknown>
   4:   0x8bfd83 - <unknown>
   5:   0x5d300b - <unknown>
   6:   0x64c91f - <unknown>
   7:   0x8a0e7b - <unknown>
   8:   0x89f8db - <unknown>
   9:   0x8a45b7 - <unknown>
  10:   0x4bcd93 - <unknown>
  11:   0x4ad973 - <unknown>
  12:   0x4bd8e3 - <unknown>
  13:   0x8bfa2b - <unknown>
  14:   0x8e665b - <unknown>
  15:   0x8c6ce3 - <unknown>
  16:   0x4ae453 - main
  17: 0xb6cc1677 - __libc_start_main`

krysztalzg, 2018-05-10:
Bumbing this, it was working fine for first 2-3 weeks but lately its crashing like this all time.
```
raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2018-05-10 10:04:52 UTC; 5min ago
  Process: 1021 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 1018 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 1024 (librespot)
   CGroup: /system.slice/raspotify.service
           └─1024 /usr/bin/librespot --name raspotify --backend alsa --bitrate 320 --disable-audio-cache --enable-volume-normalisation --linear-volume --initial-volume=100 --username --password --device hw:1

May 10 10:09:40 PiZeroKrysztal librespot[1024]:   10:   0x50785f - librespot_playback::player::PlayerInternal::run::h296c59bd7682ffbf
May 10 10:09:40 PiZeroKrysztal librespot[1024]:   11:   0x4874bb - std::sys_common::backtrace::__rust_begin_short_backtrace::h0ef34e9728d818f7
May 10 10:09:40 PiZeroKrysztal librespot[1024]:   12:   0x49aa03 - std::panicking::try::do_call::h55970923d54b3006
May 10 10:09:40 PiZeroKrysztal librespot[1024]:   13:   0x96f673 - __rust_maybe_catch_panic
May 10 10:09:40 PiZeroKrysztal librespot[1024]:                        at libpanic_unwind/lib.rs:102
May 10 10:09:40 PiZeroKrysztal librespot[1024]:   14:   0x49c92b - <F as alloc::boxed::FnBox<A>>::call_box::h6c97a59281f0cee4
May 10 10:09:40 PiZeroKrysztal librespot[1024]:   15:   0x9662df - std::sys::unix::thread::Thread::new::thread_start::hb1d1727b173caaf8
May 10 10:09:40 PiZeroKrysztal librespot[1024]:                        at /checkout/src/liballoc/boxed.rs:798
May 10 10:09:40 PiZeroKrysztal librespot[1024]:                        at libstd/sys_common/thread.rs:24
May 10 10:09:40 PiZeroKrysztal librespot[1024]:                        at libstd/sys/unix/thread.rs:90
```

andy646362, 2018-05-10:
Same issue here, log below:

`### May 10 14:09:05 osmc librespot[1913]:   10:   0x57485f - librespot_playback::player::PlayerInternal::run::h296c59bd7682ffbf
May 10 14:09:05 osmc librespot[1913]:   11:   0x4f44bb - std::sys_common::backtrace::__rust_begin_short_backtrace::h0ef34e9728d818f7
May 10 14:09:05 osmc librespot[1913]:   12:   0x507a03 - std::panicking::try::do_call::h55970923d54b3006
May 10 14:09:05 osmc librespot[1913]:   13:   0x9dc673 - __rust_maybe_catch_panic
May 10 14:09:06 osmc librespot[1913]:                        at libpanic_unwind/lib.rs:102
May 10 14:09:06 osmc librespot[1913]:   14:   0x50992b - <F as alloc::boxed::FnBox<A>>::call_box::h6c97a59281f0cee4
May 10 14:09:06 osmc librespot[1913]:   15:   0x9d32df - std::sys::unix::thread::Thread::new::thread_start::hb1d1727b173caaf8
May 10 14:09:06 osmc librespot[1913]:                        at /checkout/src/liballoc/boxed.rs:798
May 10 14:09:06 osmc librespot[1913]:                        at libstd/sys_common/thread.rs:24
May 10 14:09:06 osmc librespot[1913]:                        at libstd/sys/unix/thread.rs:90
~
`
After the song ends and connection is lost:
`May 10 14:13:57 osmc systemd[1]: raspotify.service: Main process exited, code=exited, status=101/n/a
May 10 14:13:57 osmc systemd[1]: raspotify.service: Unit entered failed state.
May 10 14:13:57 osmc systemd[1]: raspotify.service: Failed with result 'exit-code'.
`

rodet, 2018-05-12:
FYI, there have been similar issues opened to the dependency librespot https://github.com/librespot-org/librespot/issues/215 .

closed 2021-11-24 by github-actions[bot]