dtcooper/raspotify #85: Cannot connect to raspotify on newest version

l3d00m, 2018-02-23:
Hello, I've just updaed to 0.10.3 from 0.10.1 and I'm unable to connect to raspotify. It is displayed as a device and shows "Connecting..." when attempting to connect, but it does not connect. The service log does not show any info.

tejp, 2018-02-23:
I got the same problem on 0.10.3, upgraded from 0.9.1.

l3d00m, 2018-02-23:
Ok, I just downgraded to an older version of raspotify and this issue does persist. I did an `apt-get update` on my Pi though which I haven't for some time, so it might be another package that was updated?

dtcooper, 2018-02-23:
Thanks for the report. You can downgrade for now by downloading the older packages on the Release page. Looking into this now.

dtcooper, 2018-02-23:
I'm unable to reproduce. Can you please post your `/etc/default/raspotify` file along with your `sudo systemctl status raspotify` output. (With your password censored of course.)

tejp, 2018-02-23:
I'm on first raspberry pi b. Like @l3d00m I did an `sudo apt-get update && sudo apt-get upgrade`. Raspotify is selectable but not connectable from both macOS and android clients. I will try to downgrade.

These are the upgraded packages:
```
2018-02-23 17:26:06 upgrade curl:armhf 7.52.1-5+deb9u3 7.52.1-5+deb9u4
2018-02-23 17:26:08 upgrade libcurl3:armhf 7.52.1-5+deb9u3 7.52.1-5+deb9u4
2018-02-23 17:27:50 upgrade libxml2:armhf 2.9.4+dfsg1-2.2+deb9u1 2.9.4+dfsg1-2.2+deb9u2
2018-02-23 17:27:52 upgrade bind9-host:armhf 1:9.10.3.dfsg.P4-12.3+deb9u3 1:9.10.3.dfsg.P4-12.3+deb9u4
2018-02-23 17:27:53 upgrade libisc160:armhf 1:9.10.3.dfsg.P4-12.3+deb9u3 1:9.10.3.dfsg.P4-12.3+deb9u4
2018-02-23 17:27:55 upgrade libdns162:armhf 1:9.10.3.dfsg.P4-12.3+deb9u3 1:9.10.3.dfsg.P4-12.3+deb9u4
2018-02-23 17:27:58 upgrade libisccc140:armhf 1:9.10.3.dfsg.P4-12.3+deb9u3 1:9.10.3.dfsg.P4-12.3+deb9u4
2018-02-23 17:27:58 upgrade libisccfg140:armhf 1:9.10.3.dfsg.P4-12.3+deb9u3 1:9.10.3.dfsg.P4-12.3+deb9u4
2018-02-23 17:27:59 upgrade libbind9-140:armhf 1:9.10.3.dfsg.P4-12.3+deb9u3 1:9.10.3.dfsg.P4-12.3+deb9u4
2018-02-23 17:28:00 upgrade liblwres141:armhf 1:9.10.3.dfsg.P4-12.3+deb9u3 1:9.10.3.dfsg.P4-12.3+deb9u4
2018-02-23 17:28:01 upgrade libtasn1-6:armhf 4.10-1.1 4.10-1.1+deb9u1
2018-02-23 17:28:01 upgrade firmware-atheros:all 1:20161130-3+rpi2 1:20161130-3+rpt1
2018-02-23 17:28:10 upgrade firmware-brcm80211:all 1:20161130-3+rpi2 1:20161130-3+rpt1
2018-02-23 17:28:26 upgrade firmware-libertas:all 1:20161130-3+rpi2 1:20161130-3+rpt1
2018-02-23 17:28:32 upgrade firmware-misc-nonfree:all 1:20161130-3+rpi2 1:20161130-3+rpt1
2018-02-23 17:28:41 upgrade firmware-realtek:all 1:20161130-3+rpi2 1:20161130-3+rpt1
2018-02-23 17:28:43 upgrade libavutil55:armhf 7:3.2.9-1~deb9u1 7:3.2.10-1~deb9u1+rpt1
2018-02-23 17:28:43 upgrade libswresample2:armhf 7:3.2.9-1~deb9u1 7:3.2.10-1~deb9u1+rpt1
2018-02-23 17:28:44 upgrade libvorbisenc2:armhf 1.3.5-4 1.3.5-4+deb9u1
2018-02-23 17:28:44 upgrade libvorbisfile3:armhf 1.3.5-4 1.3.5-4+deb9u1
2018-02-23 17:28:45 upgrade libvorbis0a:armhf 1.3.5-4 1.3.5-4+deb9u1
2018-02-23 17:28:45 upgrade libavcodec57:armhf 7:3.2.9-1~deb9u1 7:3.2.10-1~deb9u1+rpt1
2018-02-23 17:28:53 upgrade libavformat57:armhf 7:3.2.9-1~deb9u1 7:3.2.10-1~deb9u1+rpt1
2018-02-23 17:28:56 upgrade libcurl3-gnutls:armhf 7.52.1-5+deb9u3 7.52.1-5+deb9u4
2018-02-23 17:29:05 upgrade libisc-export160:armhf 1:9.10.3.dfsg.P4-12.3+deb9u3 1:9.10.3.dfsg.P4-12.3+deb9u4
2018-02-23 17:29:06 upgrade libdns-export162:armhf 1:9.10.3.dfsg.P4-12.3+deb9u3 1:9.10.3.dfsg.P4-12.3+deb9u4
2018-02-23 17:29:08 upgrade pi-bluetooth:armhf 0.1.6 0.1.7
2018-02-23 17:29:12 upgrade raspotify:armhf 0.9.1~librespot.20171005T125424Z.8971d3a 0.10.3~librespot.20180222T153032Z.685fb4e
```

`sudo systemctl status raspotify`:
```
● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2018-02-23 18:18:02 CET; 12min ago
  Process: 972 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 969 ExecStartPre=/bin/mkdir -p 0755 /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 975 (librespot)
   CGroup: /system.slice/raspotify.service
           └─975 /usr/bin/librespot --name raspotify (Raypee) --backend alsa --bitrate 320 --disable-audio-cache

Feb 23 18:18:01 Raypee systemd[1]: Starting Raspotify...
Feb 23 18:18:02 Raypee systemd[1]: Started Raspotify.
Feb 23 18:18:02 Raypee librespot[975]: INFO:librespot: librespot (raspotify) 685fb4e (2018-02-22). Built on 2018-02-23. Build ID: zwr5e2l
Feb 23 18:18:26 Raypee librespot[975]: INFO:librespot_core::session: Connecting to AP "gew1-accesspoint-b-bcrp.ap.spotify.com:4070"
Feb 23 18:18:26 Raypee librespot[975]: INFO:librespot_core::session: Authenticated as "xxx" !
Feb 23 18:18:26 Raypee librespot[975]: INFO:librespot_core::session: Country: "XX"
Feb 23 18:18:26 Raypee librespot[975]: INFO:librespot_playback::audio_backend::alsa: Using alsa sink
```

`cat /etc/default/raspotify`
```
# /etc/default/raspotify -- Arguments for librespot

# Device name on Spotify Connect
#DEVICE_NAME="raspotify"

# Bitrate, one of 96 (low quality), 160 (default quality), or 320 (high quality)
BITRATE="320"

# Additional command line arguments for librespot can be set below.
# See `librespot -h` for more info.
#
# To make your device visible on Spotify Connect across the Internet add your
# username and password which can be set via "Set device password", on your
# account settings, use `--username` and `--password`.
#
# To choose a different output device (ie a USB audio dongle or HDMI audio out),
# use `--device` with something like `--device hw:0,1`. Your mileage may vary.
#
#OPTIONS="--username <USERNAME> --password <PASSWORD>"

# Uncomment to use a cache for downloaded audio files. It's best to leave this
# as-is since permissions are properly set on directory `/var/cache/raspotify'.
#CACHE_ARGS="--cache /var/cache/raspotify"
```

l3d00m, 2018-02-23:
Huch, I just tried it again and it worked, I cannot reproduce the issue any longer. Only thing I did was another restart (I restarted after every update before as well) and it worked like a charm. I also upgraded to 0.10.3 and it works now, I have no idea what the issue was.

I'm using the default config (as I uninstalled everything to pinpoint the issue) and the status looked exactly like @tejp's.

tejp, 2018-02-23:
I just fired up the Spotify client again and it works. Didn't do anything.

Thanks for developing raspotify!

ZM71, 2025-01-27:
I'm unable to connect without premium from android phone but other devices like Chromecast and Tv Box works fine please help me, Raspberry Pi 4.

dtcooper, 2025-01-27:
> I'm unable to connect without premium from android phone but other devices like Chromecast and Tv Box works fine please help me, Raspberry Pi 4.

Please create a new issue with a more descriptive report.

ZM71, 2025-01-27:
> > I'm unable to connect without premium from android phone but other devices like Chromecast and Tv Box works fine please help me, Raspberry Pi 4.
> 
> Please create a new issue with a more descriptive report.

No output check the link for picture

ZM71, 2025-01-27:
> > I'm unable to connect without premium from android phone but other devices like Chromecast and Tv Box works fine please help me, Raspberry Pi 4.
> 
> Please create a new issue with a more descriptive report.

I already have an issue opened.

dtcooper, 2025-01-27:
@l3d00m 🙄