dtcooper/raspotify #87: Can no longer connect to device

Powana, 2018-02-27:
I installed the package a few days ago and it worked fine for a day or two. However today when I attempt to reconnecting to my Pi through either my phone (android) or the desktop client, the device is shown in the list of available devices, but selecting it does nothing on PC and gets stuck on "connecting" on my phone.

I haven't changed any settings in ```/etc/default/raspotify``` apart from the device name, and uncommenting or changing anything in the  "OPTIONS" option causes the service to stop working. 

I've been messing around trying to fix this for a while now but I can't seem to get it to work, any help is greatly appreciated.

```sudo systemctl status raspotify```:
```raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2018-02-27 20:28:46 CET; 7min ago
  Process: 895 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 892 ExecStartPre=/bin/mkdir -p 0755 /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 898 (librespot)
   CGroup: /system.slice/raspotify.service
           └─898 /usr/bin/librespot --name Blueberry Pi --backend alsa --bitrate 160 --disable-audio-cache

Feb 27 20:28:46 blueberrypi systemd[1]: Starting Raspotify...
Feb 27 20:28:46 blueberrypi systemd[1]: Started Raspotify.
Feb 27 20:28:46 blueberrypi librespot[898]: INFO:librespot: librespot (raspotify) 4f3a594 (2018-02-17). Built on 2018-02-20. Build ID: 8u37iq02
```

PlanetWalker, 2018-02-28:
Had the same Problem.
For me a reinstall worked just fine. 

PlanetWalker

Death111, 2018-03-03:
had the same problem for some days. I issued "sudo apt-get install raspotify" and it updated and a "sudo systemctl restart raspotify" to restart the service - working again 👍

Powana, 2018-03-03:
Reinstalling did seem to fix the issue, I wonder what caused it.

closed 2018-03-03 by Powana